realsenseai/librealsense #5614: T265 ability to open pipeline using serial number

orshefi, 2020-01-09:
Hi Guys,

I'm testing #5213 to see rather all functionalities previously used by my system remains.
So far I found out that I can no longer open T265 device by serial number with enable_device api.

Here is a simple init function to check which devices & their serial number are connected, and try to open them with pipe API.
```
rs2::context ctx;
    auto devices = ctx.query_devices();
    for(auto dev : devices){
        std::cout << dev.get_info(RS2_CAMERA_INFO_NAME) << std::endl;
        std::cout << dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) << std::endl;
    }
    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    // Add pose stream
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
    std::string serial = "0000909212111581";
    cfg.enable_device(serial);
    // Start pipeline with chosen configuration
    pipe.start(cfg);
```

Here is the output:
```
Intel RealSense T265
0000909212111581
RealSense error calling rs2_pipeline_start_with_config(pipe:0x186a580, config:0x185b600):
    No device connected
```

As you can see:

1. Serial number is padded with zeros - new behavior, I'm using a certain convention in my config file that is broken according to this change. (I had also tried without the padding to open the device serial - 909212111581)
2. Device is enumerated but not opened with the serial number listed
3. I also run rs-pose example and the camera was able to open and stream data with pipe API (just without the enable device API).

radfordi, 2020-01-09:
Thanks for your report, @orshefi.  I just noticed this myself, and I was wonder if this used to be the case or not.  I'll put together a fix.

JeremyBYU, 2020-01-09:
Having the same issue. I had to open the t265 individually by the sensor using `sensor.open(stream_profile)`

zainmehdi, 2020-01-31:
I tried this fix but still having the same error. Any updates on it ?

radfordi, 2020-01-31:
Hi @zainmehdi, there are two fixes available in ##5615 and #5645.

zainmehdi, 2020-02-03:
@radfordi Thanks alot for your reply. i already tried #5615 and it didnt work for me. I will try 5645 and update.

orshefi, 2020-02-05:
Hi @radfordi ,

Just tested the latest development version with #5645 merged into.
The serial number indeed is in the same convention as libtm and its great.
But nonetheless, the pipe cannot start with the serial number enabled in the config.

radfordi, 2020-02-05:
Hi @orshefi, the current `development` version (d363eb9) with #5645 is still missing #5615, so you'll need to merge that before it'll work.

Andrew-fedoralabs, 2020-02-16:
Hello @radfordi,
I have pulled the dev branch, merged #5615 and built. 
I am no longer getting"no device connected" but am now getting:
 "RealSense error calling rs2_pipeline_start_with_config(pipe:0x7f6c1404e7c0, config:0x7f6c1404be00):
    Unable to open device interface"

I have tried to run rs-multicam and it now fails with the T265 ( it was working previously with master branch, but my application was not able to open the device with serial numaber and I was getting "no device connected" leading me to find this potential fix)

Any suggestions? 

Thanks.

Andrew-fedoralabs, 2020-02-16:
I have Rolled back to v2.31.0 but it still has the "no device connected" Issue. Does anyone know what version endable_device by serial number was working with T265? Its required for my application to continue development. Thanks.

zwn, 2020-02-16:
I am successfully working with v2.31.0 and v2.29.0. Please check, if you are not hitting "the other problem" when T265 is not detected at all right after boot.

Andrew-fedoralabs, 2020-02-16:
Hi Thank you for the response. This is interesting because im getting the same result on 2.31.0 that I was having before, It works if I do not start a pipe with a context or call enable_device(). Example:

This will work:
(pipe is defined earlier in the code)
rs2::config cfg;
pipe = rs2::pipeline();
cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
cfg.enable_stream(RS2_STREAM_FISHEYE, 1, RS2_FORMAT_Y8);
cfg.enable_stream(RS2_STREAM_FISHEYE, 2, RS2_FORMAT_Y8);
pipe.start(cfg);

This will not work and fail with no device connected:
(dev was queried from context earlier in code)
rs2::config cfg;
pipe = rs2::pipeline(context);
cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
cfg.enable_stream(RS2_STREAM_FISHEYE, 1, RS2_FORMAT_Y8);
cfg.enable_stream(RS2_STREAM_FISHEYE, 2, RS2_FORMAT_Y8);
pipe.start(cfg);

Is this how you are configuring the device with serial number and pipe?

zwn, 2020-02-17:
I am using python and this little script runs without errors for me with v2.29. I am not at the position to test it with 2.31 as my other T265 is in my luggage that got hold up by the airline.

```python
ctx = rs.context()
pose_pipeline = rs.pipeline(ctx)
pose_cfg = rs.config()
pose_cfg.enable_device(serial)
pose_cfg.enable_stream(rs.stream.pose)
pose_pipeline.start(pose_cfg)
pose_pipeline.stop()
```

radfordi, 2020-02-17:
@Andrew-fedoralabs, this issue should be fixed by #5615 which hasn't been merged yet.  Because you are mentioning `FISHEYE`, you can expect only a T265 to satisfy your `config`, so unless you have two T265s, you shouldn't need the serial number.

vdutell, 2020-02-23:
Thanks developers for all your work on this. I am having the same issue - and am unable to do the kernel path to build myself for other unrelated reasons.... Any idea when this merge will happen? Thanks!

Microset, 2020-04-06:
Hi,
For me T265 ability to open pipeline using serial number in python3.5 not working in multiple camera modes  from v2.31 till now for v2.34.  I am successfully working with  v2.29. When we can hope to correct this bag?

ivomarvan, 2020-04-22:
Same like Microset.

ivomarvan, 2020-04-22:
For example, this error is reflected in rs-enumerate-devices utility. Try it with the camera connected and the -d parameter.

rs-enumerate-devices -d

Result:
....
Default streams:
RealSense error calling rs2_config_resolve(config:0x559d0b529d00, pipe:0x559d0b547540):
    No device connected

In the source code it is called: 
cfg.enable_device (dev.get_info (RS2_CAMERA_INFO_SERIAL_NUMBER));

dorodnic, 2020-07-06:
Should be resolved in 2.35.2

PR #5615 (2020-06-01): Support filtering USB devices (T265's) by Serial Number
Fixes #5614.
closed 2020-07-06 by dorodnic