realsenseai/librealsense #5616: How to get cloud data instead of Mesh in export_to_ply() ?

vishnukumarts, 2020-01-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Windows 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python }                          | 
| Segment			|  others }                          | 

### Issue Description
I used export_to_ply API function provided in the below link,

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1points.html#a803e6124fb393f956e9cc5dfc0174784

I am getting the nice completed surface, I just want to understand instead of providing points as output, did Intel do meshing over the pointcloud data?  If yes, what kind of meshing.

Also, what is API I have use if require only point cloud without mesh?

RealSenseCustomerSupport, 2020-01-21:
----------------------------------------------

@vishnukumarts Please check if "OPTION_PLY_BINARY" listed in PR below can meet your requirement or not. Thanks!

https://github.com/IntelRealSense/librealsense/pull/4906

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@vishnukumarts Did you get chance to try "OPTION_PLY_BINARY"? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@vishnukumarts Could you please update? If we don’t hear from you in 7 days, this issue will be closed. Thanks!

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@vishnukumarts Sorry we didn't hear from your for weeks. Will close the ticket at this point. Please feel free to create another ticket if you still have questions or issues.  Thanks!

closed 2020-02-17 by RealSenseCustomerSupport