realsenseai/librealsense #5617: RealSense Android SDK Java

guptaabhinav1512, 2020-01-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Android |
| SDK Version                            |  2.22.0                           | 
| Language                            |  Java                          | 
| Segment			|  Smartphone                          | 

### Issue Description
Hi, I am trying to use Realsense D400 series with my Android device using com.intel.realsense:librealsense sdk. I had few general questions :

1. I am using enableAllStreams(), Is there a way I can set the FPS rate for all the streams instead of using enableStream()?
2. I am using composite FrameSet, does the size() of the Frameset denote the FPS? Or are these two different? For example: if 30 FPS is the streaming rate, does each FrameSet contain 30 Frames?
3. How do I find out the current streaming FPS rate on an Android device in RealTime? How can I change this without setting enableStream()? I want to understand what's my FPS while using the application.

kafan1986, 2020-01-10:
1) From the constructor, no does not look it supports custom FPS for enableAllStreams. Not sure what is the benefit of using enableAllStreams vs enableStream? May be couple of lines of code.
2) There is no method of size() but did see there is getSize() which returns number of frames. So if you have enabled IR feed (left IR camera), Depth Feed and RGB Feed, then getSize will return 3 i.e. 3 frames count. It depends on how you have enabled streaming callback or self polling (either with or without wait). If it is callback then the callback method will be called at the same rate as FPS.
3) Current streaming FPS rate can be little different from what you have set. You need to calculate it yourself by checking the time difference between each subsequent frameset callback. Also, it is better to calculate the FPS for each individual stream by checking the respective frame number and  comparing it with last corresponding frame number. It is already implemented in the test android application ("camera" by MatKatz).

guptaabhinav1512, 2020-01-11:
Thanks for the response, it was really helpful. One last question, so If the callback is being called as per the FPS, if I add delay to this. I can technically change the FPS to anything lower than 6 FPS right?

closed 2020-01-12 by guptaabhinav1512

kafan1986, 2020-01-12:
> Thanks for the response, it was really helpful. One last question, so If the callback is being called as per the FPS, if I add delay to this. I can technically change the FPS to anything lower than 6 FPS right?

Callback gives you access to the current frameset. The FPS at the camera will remain the same that you setup using config i.e. frameset will get dropped if you are currently not processing them.