realsenseai/librealsense #5618: Build configuration options used in official PPA builds

WasabiFan, 2020-01-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | N/A | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Ubuntu 18.04 arm64  | 
| Kernel Version (Linux Only)    | N/A | 
| Platform                                 | Nvidia Jetson Xavier  |
| SDK Version                            |  All | 
| Language                            | All                          | 
| Segment			|  N/A                          | 

### Issue Description

What CMake flags are used when building each of the `.deb` packages at `http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo`? Where can I find them?

In particular, this page (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) suggests at the bottom building with `FORCE_LIBUVC=true` and `BUILD_WITH_CUDA=true` if building from source. But up above it recommends using the prebuilt packages. So are builds for this architecture built with those flags? Or is the page not internally consistent?

ev-mp, 2020-01-12:
@WasabiFan hello, the Debian packages for Arm64 are Jetson-optimized, i.e. built with `FORCE_LIBUVC=true` and `BUILD_WITH_CUDA=true` as recommended.

WasabiFan, 2020-01-15:
@ev-mp It looks like that build configuration doesn't include the Python wrapper. The wheels published on PyPi don't include arm64 builds. How am I supposed to install the Python wrapper?

ev-mp, 2020-01-15:
That's correct, currently the `pip install` packages are generated for X86/X64 architecture only.
For Arm64 you need to build the python wrapper using the [installation guide](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source)

WasabiFan, 2020-01-15:
The issue is that you can't build the Python wrapper without building all of librealsense, as far as I can tell. So the pre-built packages are only useful if you're writing in C++ — if you want to write in Python or another language, you have to compile from source anyway, right?

Can you provide or point me to the exact script and build options that are used for official builds so I can replicate it as closely as possible?

ev-mp, 2020-01-16:
The rule used to configure the SDK for Jetson debians is :
>override_dh_auto_configure: dh_auto_configure -- -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DBUILD_WITH_CUDA:BOOL=ON -DFORCE_RSUSB_BACKEND:BOOL=ON

RealSenseCustomerSupport, 2020-01-24:
----------------------------------------------

Hi @WasabiFan,

Has the comment provided by @ev-mp helped to find an answer to your question? Please let us know if you need any further assistance with this matter.

Thank you!

WasabiFan, 2020-01-24:
It was helpful. @ev-mp where can this information be found publicly? Can you publish the official build scripts?

ev-mp, 2020-02-04:
@WasabiFan, as for the inquiry -
while currently it is not in the public domain upstreaming the SDK debian distribution is one of the established team's goals, which in turn will automatically open-source the repositories.
Unfortunately I cannot elaborate on scheduling.

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

@WasabiFan,

Do you need any additional clarifications regarding this topic?

WasabiFan, 2020-02-09:
I don't need clarification, but this issue should presumably stay open until the build script configurations are publicly available.

RealSenseSupport, 2020-06-08:
@WasabiFan 

If no further clarifications are needed, this thread will be closed since we already have an open issue where updates about the progress on "making build script configurations publicly available" will be published: https://github.com/IntelRealSense/librealsense/issues/4431

Please don't hesitate to open a new thread if you have additional questions!
Thank you!

closed 2020-06-08 by RealSenseSupport