realsenseai/librealsense #5620: tare calibration API

eliranno, 2020-01-10:
im trying to get the tare calibration to work. im using rs2_run_tare_calibration API with latest firmware and sdk;however, I always get calibration table which points to NULL. I was wondering if anyone tried this API and whether im missing some steps here. 
according to the error::handle function one or more parameters are invalid. please note that im using default json by passing nullptr. same for callback functions

``
std::future<const rs2_raw_data_buffer*> RealSenseDevice::startCalibration(const float ground_truth_mm, const void *jsonContent,int contentSize,rs2_update_progress_callback_ptr progressCallback, const int timeout) {

auto future = std::async(std::launch::async,[this,ground_truth_mm,jsonContent,contentSize,progressCallback,timeout]()->const rs2_raw_data_buffer*{

rs2_error* e = nullptr;

auto calibrationTable = rs2_run_tare_calibration(device.get().get(),20000,nullptr,0,nullptr,nullptr,timeout*10,&e);
rs2::error::handle(e);

if(calibrationTable!= nullptr){

Log::Debug("SUCCESSSSSSSSSSSSSSSSSS!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!") << Log::Endl;

} else

Log::Debug("FAILEDDDDDDDDDDDDDDDDDDDDDDDDDDDDD!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!") << Log::Endl;

return calibrationTable;

});

return future;

}

dorodnic, 2020-01-11:
Hi @eliranno 
This API is being used in the [Viewer](https://github.com/IntelRealSense/librealsense/blob/master/common/on-chip-calib.cpp#L319-L323), and, indirectly, from a python example I believe. 
You are using modern C++ in your App, why are you calling the C-API? (I'm just curious)
Are you getting any error from the call? I assume you are running it during streaming? Did you manage to run self-cal successfully?

eliranno, 2020-01-11:
Hi Surgey,

Thank you for your reply.

Yes, my application is written in modern c++. I have tried using both C API
and also cast the rs2_device to auto_calibeated_device and then call the
tare function. Both failed to execute. Is there a better way of calling it
? Also, I was told that in order to run the tare calibration the device
configuration must be set to 90fsp which I have not tried yet.

Thanks for your help

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

@eliranno What's the error details when you get execution failure? Could you please provide more logs or snapshots? Thanks!

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@eliranno Could you please provide more details if you still need support on this? Thanks!

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

@eliranno Sorry we didn't get response from your side for weeks. Will close the issue at this point. Please feel free to create new one if you still have issues or questions. Thanks!

closed 2020-03-04 by RealSenseCustomerSupport