realsenseai/librealsense #5624: Realsense D415 - check highest point in Unity wrapper every frame

ASinha92, 2020-01-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version |   Win10  |
| Platform                                 | PC |
| SDK Version                            |  2.0                         | 
| Language                            |  unity                   | 

### Issue Description
Hello, Im trying to find out if the height detected by my two realsense caemras (D415) is more than a threshhold value i set.

I cannot find a variable i can access from the scripts provided. I am very new to unity and would like to test out if this is a feasible solution for what I want to achieve.

Any suggestions on which scene or scripts to get this height value out of? Basically want to see if a person walks into the frame. my cameras are on the ceiling looking at the ground. 
I have gone through startup tutorial.

MartyG-RealSense, 2020-01-10:
Does the distance measuring script in the link below meet your needs, please?

https://github.com/IntelRealSense/librealsense/issues/4201

ASinha92, 2020-01-10:
hi Marty, 
the script would create a 2D array every frame.. also where do i insert that n call it? wouldnt it be overkill to just detect a change in height ie some1 walking into frame? i do not need to measure in a certain unit. I was looking into the AR background script. can that be used to do what i need?

MartyG-RealSense, 2020-01-10:
Someone else had the same idea as you about ARBackground, and it is discussed in detail in the link below.  The old forum it is on is being removed on January 13 though, so please store the page's information somewhere safe such as on a print-out or a wordprocessor doc if you want to keep access to it.

https://forums.intel.com/s/question/0D50P0000490LvdSAE/get-depth-data-from-d415-within-a-unity-script?language=en_US

MartyG-RealSense, 2020-01-11:
I saved the linked-to page as a PDf file and attached it here in case you did not get to visit the forum before it closed down.

[Get depth data from D415 within a Unity script.pdf](https://github.com/IntelRealSense/librealsense/files/4049133/Get.depth.data.from.D415.within.a.Unity.script.pdf)

closed 2020-02-03 by RealSenseCustomerSupport