realsenseai/librealsense #5630: IR camera doesn't work well in darkness

HIAI2019, 2020-01-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10) | 
| Platform                                 | PC  |
| SDK Version                            |   legacy / 2.0                           | 
| Language                            |  C++/opencv                         | 

### Issue Description

I use D435 in darkness and close laser,there is no difference between IR camera and RGB camera.
infrared image：
[image]

my codes:
`int main(int argc, char * argv[])
{
	int width = 1280;
	int height = 720;
	int fps = 30;
	
	
	rs2::config config;
	config.enable_stream(RS2_STREAM_INFRARED, 1, width, height, RS2_FORMAT_Y8, fps);
	config.enable_stream(RS2_STREAM_INFRARED, 2, width, height, RS2_FORMAT_Y8, fps);

	// start pipeline
	rs2::pipeline pipeline;
	rs2::pipeline_profile pipeline_profile = pipeline.start(config);
	rs2::device selected_device = pipeline_profile.get_device();
	auto depth_sensor = selected_device.first<rs2::depth_sensor>();

	depth_sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 0.f);
	while (1) // Application still alive?
	{
		// wait for frames and get frameset
		rs2::frameset frameset = pipeline.wait_for_frames();

		// get single infrared frame from frameset
		//rs2::video_frame ir_frame = frameset.get_infrared_frame();

		rs2::video_frame ir_frame_left = frameset.get_infrared_frame(1);
		rs2::video_frame ir_frame_right = frameset.get_infrared_frame(2);

		cv::Mat dMat_left = cv::Mat(cv::Size(width, height), CV_8UC1, (void*)ir_frame_left.get_data());
		cv::Mat dMat_right = cv::Mat(cv::Size(width, height), CV_8UC1, (void*)ir_frame_right.get_data());
		cv::imshow("img_l", dMat_left);
		cv::imshow("img_r", dMat_right);
		char c = cv::waitKey(1);
		if (c == 's')
		{

		}
		else if (c == 'q')
			break;
	}

	return EXIT_SUCCESS;
}`

dorodnic, 2020-01-13:
IR cameras do not "see in the dark". You need some source of light (in case of infrared cameras, infrared light) to see something... 
One source you can use is the camera projector. Alternatively, you can use an IR-LED (this is what most face authentication systems are doing)
Alternatively, you may want to disable auto-exposure and crank-up both exposure and gain to amplify any ambient light available.

HIAI2019, 2020-01-14:
> IR cameras do not "see in the dark". You need some source of light (in case of infrared cameras, infrared light) to see something...
> One source you can use is the camera projector. Alternatively, [...]

Thanks.One more question,In order to choose a approprite IR-LED ,I want to know the focal length of d435 IR camera .

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

@HIAI2019 The focal length is 1.93mm for D435 IR. For details you can check the datasheet in https://www.intelrealsense.com/wp-content/uploads/2019/10/Intel-RealSense-D400-Series-Datasheet-Oct-2019.pdf

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@HIAI2019 Any other questions from your side? Looking forward to your reply. Thanks!

closed 2020-02-17 by dorodnic