realsenseai/librealsense #5631: D435i not recognized, Ubuntu 18.04.03, Realsense 2.31.0

limap, 2020-01-13:
Hi,

the D435i is not recognized by the realsense-viewer. 

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 2.31.0| 
| Operating System & Version |   Ubuntu 18.04.3 LTS | 
| Kernel Version (Linux Only)    |  4.15.0-74-generic                                        | 
| Platform                                 | PC |

### Issue Description
I installed librealsense2-dkms, librealsense2-utils, as well as  librealsense2-dev and librealsense2-dbg. But when opening the realsense viewer it says "Connect a RealSense Camera or Add Source".

modinfo uvcvideo | grep "version:" returned 
`version:        1.1.2.realsense-1.3.10`
`srcversion:     B1E013744214D3216F199D4`

rs-fw-update -l returns
`there are no connected devices`

lsusb returns
`Bus 002 Device 002: ID 8087:8000 Intel Corp. `
`Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 001 Device 002: ID 8087:8008 Intel Corp. `
`Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 004 Device 006: ID 8086:0b3a Intel Corp. `
`Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 003 Device 011: ID 413c:301a Dell Computer Corp. `
`Bus 003 Device 010: ID 413c:2113 Dell Computer Corp. `
`Bus 003 Device 009: ID 0d8c:013c C-Media Electronics, Inc. CM108 Audio Controller`
`Bus 003 Device 008: ID 05e3:0608 Genesys Logic, Inc. Hub`
`Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

and dmesg answer includes
`[  673.736093] usb 4-1: USB disconnect, device number 2`
`[  685.304126] usb 4-1: new SuperSpeed USB device number 3 using xhci_hcd`
`[  685.324709] usb 4-1: New USB device found, idVendor=8086, idProduct=0b3a`
`[  685.324711] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3`
`[  685.324712] usb 4-1: Product: Intel(R) RealSense(TM) Depth Camera 435i`
`[  685.324714] usb 4-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i`
`[  685.324715] usb 4-1: SerialNumber: 934123053520`

ev-mp, 2020-01-14:
@liman hello, can you provide the full kernel log ?
Additionally please inspect the Viewer's log after setting the log verbosity to "Debug"

limap, 2020-01-16:
Solved. The issue was that Secure boot was enabled, so librealsense2-dkms kernel rules could not be applied. Works fine after disabling secure boot.

closed 2020-01-16 by limap

jarvis-owl, 2020-02-18:
Had this problem after Ubuntu update; upgrade and some weird MOK bios stuff (idk).
`Linux 5.3.0-40-generic #32~18.04.1-Ubuntu SMP x86_64 x86_64 x86_64 GNU/Linux`
`D435`