realsenseai/librealsense #5634: all the 3D points without using rs.rs2_deproject_pixel_to_point() fuction

scizors, 2020-01-13:
camera: Intelrealsense D435 
 code: python
   frames = pipe.wait_for_frames()
    time.sleep(10)
    # Fetch color and depth frames
    depth = frames.get_depth_frame()
    print(depth)
    color = frames.get_color_frame()
    print(color)
    print(type(color))

    # Tell pointcloud object to map to this color frame
    pc.map_to(color)

    # Generate the pointcloud and texture mappings
    points = pc.calculate(depth)
    print(points)
    vtx = np.asanyarray(points.get_vertices())
    tex = np.asanyarray(points.get_texture_coordinates()) 

how to get all the 3Dpoints form the above code without using
 rs.rs2_deproject_pixel_to_point(intrinsics,[y,x],depth_value_z)

ev-mp, 2020-01-16:
@scizors hello, in the above code you do call for 2D->3D  transformation ((rs2_deproject_pixel_to_point)) by running the following line
>points = pc.calculate(depth)

And then you read the 3D points with
> vtx = np.asanyarray(points.get_vertices())

So I'm not sure I understand the question. Can you be more specific?

scizors, 2020-01-16:
rs.rs2_deproject_pixel_to_point(intrinsics,[y,x],depth_value_z) 
by using this I will get 3D point xy and z

is there and other way to get 3D points of all the pixel by using the below code(generation of ply)
I think ply also contain 3D points

pc.map_to(color)
# Generate the pointcloud and texture mappings
points = pc.calculate(depth)
print(points)
vtx = np.asanyarray(points.get_vertices())
tex = np.asanyarray(points.get_texture_coordinates()

ev-mp, 2020-01-20:
@scizors , I'm sorry but still don't understand the question - can you try to explain the use-case or what you're trying to achieve ?
Exporting 3D point-clouds as PLY is the SDK-supported option for data sharing and it is available either as an on-demand feature in `realsense-viewer` or as a dedicated API call.

closed 2020-02-04 by RealSenseCustomerSupport