realsenseai/librealsense #5635: rs-pcl-color example not generating RGB texture

vftejada, 2020-01-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-74                                         | 
| Platform                                 | PC |
| SDK Version                            |  latest 2.31                           | 
| Language                            |  pcl 1.7                         | 
| Segment			|  {Robot }                          | 

### Issue Description
While executing the rs-pcl-color example the RGB texture is not generated. Checking out the output generated  by the .pcd file, it can be appreciated that RGB functionality to map color to point cloud data (XYZ) is not working as expected. 
Any suggestions to overcome this issue would be much appreciated.

[image]
[image]

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

@vftejada  Your issue might be caused by RGB calibration table crashed. Did you get chance to try rs-pointcloud https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud to see if any issue also?  Looking forward to your update. Thanks!

vftejada, 2020-02-06:
Hi,
I managed to get the the example working as expected with the RGB texture functionality by upgrading to PCL version 1.8.  However, I noticed another issue when uploading the .pcd files generated using the PCL point cloud visualization implemented in the Load_PCLFile function. The frame is always shown upside down (rotated 180 degrees around de z axis).
Any ideas as to how to solve this issue?

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

The rotation/flipping may be due to the use of different coordinate system. Please feel free to open a new issue for any other question.
 Thank you.

closed 2020-02-18 by RealSenseCustomerSupport