realsenseai/librealsense #5640: What the best way 同

princetonpku, 2020-01-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

princetonpku, 2020-01-14:
I found two way to save bag files. The first is:
```
//configure
rs2::config cfg; // Declare a new configuration
cfg.enable_record_to_file("a.bag");
cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_RGB8);
cfg.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16);
pipe->start(cfg); //File will be opened at this point
```

The second is:

`
//recorder class
rs2::recorder recorder("./all_combinations_depth_color.bag", dev);`

I want ask which is the best one?

dorodnic, 2020-01-14:
First one is the recommended, this is what is documented in the playback-record example.
Pipeline in general is a convenience wrapper around devices, so when you use the first method, under the hood it will use the second API. You are still welcomed to use it directly, but it involves more complexity in setting-up.
Third method is actually not related to bag files and it requires rs2_internal include, so you are not recommended to use it.

princetonpku, 2020-01-15:
Thanks for your quick reply! I got it.
I have test the third method, The bag file saved from it cannot be loaded by the realsense-viewer.exe. 
It only can be loaded by the following:

```
rs2::context ctx = rs2::mock_context("./test1.bag", "Sync__sanity.1.test");
std::shared_ptr<rs2::pipeline> pipe = std::make_shared<rs2::pipeline>(ctx);
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8);
cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16);
cfg.enable_device_from_file("test1.bag");
pipe->start(cfg);

while (true)
{
  rs2::frameset frames = pipe->wait_for_frames(); // wait for next set of frames from the camera
  rs2::video_frame color = frames.get_color_frame();
  if (!color)continue;
  cv::Mat img = frame_to_mat(color);
  cv::imshow("", img);
  cv::waitKey(1);
}
```

In fact, There are some other problems. when I see the frame, The frames is not in order, Some frames are repeated.

Shivani1796, 2020-01-16:
@dorodnic HI, as you said https://github.com/IntelRealSense/librealsense/issues/5640#issuecomment-574290901 , can we use same for T265 camera to save bag file.

RealSenseCustomerSupport, 2020-01-31:
----------------------------------------------

Hi @princetonpku

There seems to be a few different methods outlined.  I believe this thread can be closed out now.

closed 2020-02-25 by RealSenseCustomerSupport