realsenseai/librealsense #5642: Realsense D435i Depth Issue After Prolonged Use

Jmelendez93, 2020-01-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      | 
| Firmware Version                   | 05.12.01.00                                             | 
| Operating System & Version |   Linux (Ubuntu 18.04)                            | 
| Kernel Version (Linux Only)    | 4.15.0-72-generic                                   | 
| Platform                                 | Dell Optiplex (x86_64 architecture Edge PC)  |
| SDK Version                            | 2.30.0.1184                                             | 
| Language                            |  python                                                      | 

### Issue Description
I am using the Realsense D435i to detect when an object is flat/perpendicular to the view of the camera. The functionality is in place and works as intended when the depth information comes in correctly. However, when using the camera for an extended period of time (after 25-30 executions of the whole code, which consists of ~15 seconds of frame-grabbing), I am noticing the frame will occasionally return a series of depth images that are incredibly noisy and invalid: 

[image] [image]

When the frames are coming in just fine, they look more like this and can easily distinguish between objects in the frame:

[image] [image]

I have a custom class I created for the Realsense methods. I create an instance of this Realsense class in my code and in the class I have the pipeline, config, and profile instantiated among other properties (such as sensors enabled, filters used, etc.). I am currently using the align method as well as the spatial and hole filters in disparity mode. The current settings are 1920x1080 at 30 fps for RGB and 1280x720 at 30 fps for Depth. 

I had a colleague mention they had issues when calling the pipeline from within a separate class and that removing it and putting it in the main function resolved their issues, not sure if that may be related to this. They were getting slow frames followed by spikes of inconsistent depth.

I have checked the temperature before and after every use and the Projector temp varies between 29 and 33 while the ASIC temp varies between 34 and 39, so I don't believe it's an issue with the device getting too hot. There is an outer try-except statement around the whole code followed by finally block with a close() on the pipeline and a USB reset. There is also a hardware_reset() when there is an error while grabbing frames.

When the issue occurs, it is temporarily fixed by unplugging the device then plugging it back in (or perhaps this is really being fixed by giving the camera some downtime when I'm unplugging it), but it is more likely to occur after it occurs the first time. I am working on a project that will eventually be in a live production environment and can not have the protocol for fixing this be individuals unplugging and re-plugging in the device from the USB port.

If anyone's seen this before, any help would be appreciated. Thanks!

RealSenseCustomerSupport, 2020-01-22:
----------------------------------------------

to have more clue, please see if you can running the viewer on your ubuntu PC when issue occurs
(before running viewer, please close your app)
When viewer launched, you can see all the setting and easier to check the depth and IR image quality  and easier to config the depth setting

-when projector is on, you should able to see lots white dots shown on the IR image
-please see if when using viewer to do the  depth streaming, issue still occurs .. or gone without power cycle the camera

you can refer to below link about how to install SDK (prebuilt viewer ) on your ubuntu PC
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Jmelendez93, 2020-01-22:
Thanks for the reply,

I have turned the projector off through the line sensor.set_option(rs.option.emitter_enabled, 0)

I will check the viewer though and see if anything pops up. Will post an update as soon as I am able to recreate and test this.

closed 2020-02-13 by RealSenseCustomerSupport