realsenseai/librealsense #5644: orb-slam2 compile problem caused by librealsense

TouchDeeper, 2020-01-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | none | 
| Firmware Version                   | none | 
| Operating System & Version |  Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0-74                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.22.0                         | 
| Language                            |  c++                        | 
| Segment			|                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I install the realsense sdk in my ubuntu. Today I want to compile orb-slam2, but error occurs:
```
make[3]: *** No rule to make target '/usr/local/lib/librealsense2.so', needed by '../lib/libORB_SLAM2.so'。Stop。
CMakeFiles/Makefile2:257: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
CMakeFiles/Makefile2:158: recipe for target 'CMakeFiles/mono_kitti.dir/rule' failed
make[1]: *** [CMakeFiles/mono_kitti.dir/rule] Error 2
Makefile:144: recipe for target 'mono_kitti' failed
make: *** [mono_kitti] Error 2
```
I only add a `#include <unistd.h>` in system.h of orb-slam2. I don't know why the orb-slam2 will depend on librealsense, could you please provide some advice, thanks a lot!

TouchDeeper, 2020-01-16:
Anyone help

RealSenseSupport, 2020-12-05:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-12-05 by TouchDeeper

almondboi, 2020-12-11:
I am facing the exact same issue. Has anyone managed to solve this?

losantonopoulos, 2022-04-10:
Any solution?