realsenseai/librealsense #5646: Getting position data with D435i alone and without T265

smtabatabaie, 2020-01-15:
Hi, I have two D435i intel realsenses and wanted to get the position data of my device. Since Intel claims that you can have 6DOF data with D435i I didn't buy T265 and wanted to get position of my sensors without T265. I wanted to ask if there's a way to get this information from D435i ? I have also installed SLAM with D435i and it's working , but I can't retrieve position data from RViz.
Thanks

MartyG-RealSense, 2020-01-15:
The RealSense T265 Tracking Camera can provide pose data. The D435 and D435i do not by default though.

[image]

The official data sheet document for the 400 Series cameras states that the D435i supports 6DOF, as it has the same IMU component as the T265.  the information that can usually be accessed from the D435i's IMU is the accel and gyro.

[image]

There are workarounds for getting pose data though with OpenCV or ROS.

There was a user on the RealSense GitHub site who was using a robot arm for bin picking. Their own approach for getting pose with a D435 was an OpenCV SolvePnP algorithm.

https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-475896784

Another user offered a different approach to getting object pose with D435 and ROS.

https://github.com/NVlabs/Deep_Object_Pose/issues/10

smtabatabaie, 2020-01-15:
Thanks very much Marty,
I think there's been a misunderstanding. I want to access relative position of the RealSense sensore itself according to its environment. And as far as I searched, despite Intel claiming it has 6DOF it only has 3DOF and can only give you the rotation angles of the sensor. I'll appreciate if you can help me with this.
Thanks

MartyG-RealSense, 2020-01-15:
I apologize if I misunderstood.   When 6DOF is mentioned in terms of RealSense, it is usually in reference to the 'pose' of the device.  The full size version of the chart that I quoted about 6DOF explains that this data includes the XYZ position of the device.

[image]

RealSenseCustomerSupport, 2020-01-21:
----------------------------------------------

Thanks Marty. That is correct. The T265 provides 6DOF Pose data.

Mojtaba....will you be needing further assistance with this?

VladMVLX, 2020-01-25:
[image]
It literally says 6dof even hash tags it.

MartyG-RealSense, 2020-01-25:
Although the D435i and T265 have the same IMU component, they handle IMU data differently.  On D435i, the raw IMU data is delivered directly to the host PC.  The T265 though has a Movidius Myraid 2 Visiaul Processing Unit (VPU) chip which the D435i lacks.  This VPU can generate **pose** information by fusing inertial and video sensor data.

closed 2020-02-03 by RealSenseCustomerSupport