realsenseai/librealsense #5650: T265: Again USB problems, now with Banana PI M2 Zero

neilyoung, 2020-01-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.896 (?) |
| Operating System & Version |  ubuntu 16.04.6 LTS |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.32.0, development, 95cbea5                       | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

Well, I'm having again problems with USB, now on the a.m. platform. There is no chance to make USB work with the T265 with the master branch. Getting errors like

 

> Device-A6F8: MessageID 0x12 (DEV_START) failed with status 0x9

or

> Error: Set Start Failed (0x1)

And with the latest development branch it is not much better

> Exception  No device connected

`lsusb` does not really properly enumerate the device:

```
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 004: ID 03e7:2150  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

and  `dmesg`:

```
[ 2776.953846] usb 1-1: new high-speed USB device number 3 using ehci-platform
[ 2777.161156] usb 1-1: New USB device found, idVendor=8087, idProduct=0b37, bcdDevice=ff.ff
[ 2777.161193] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2777.161218] usb 1-1: Product: Intel(R) RealSense(TM) Tracking Camera T265
[ 2777.161239] usb 1-1: Manufacturer: Intel(R) Corporation
[ 2777.161261] usb 1-1: SerialNumber: 9291221101120000
[40832.831381] usb 1-1: USB disconnect, device number 3
[40835.202443] usb 1-1: new high-speed USB device number 4 using ehci-platform
[40835.413449] usb 1-1: New USB device found, idVendor=03e7, idProduct=2150, bcdDevice= 0.01
[40835.413519] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[40835.413565] usb 1-1: Product: Movidius MA2X5X
[40835.413608] usb 1-1: Manufacturer: Movidius Ltd.
[40835.413650] usb 1-1: SerialNumber: 03e72150

```
I know I had something running on this device already in the past, so this is maybe a regression.

What can I do in order to tackle this problem?

neilyoung, 2020-01-16:
Somebody? Any pointer? The device is not properly recognizable anymore. I think there has been a better time already :(

neilyoung, 2020-01-16:
Was trying to apply the patch as described here https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

>  Build and apply patched kernel modules for:
> * Ubuntu 14/16/18 with LTS kernel script will download, patch and build realsense-affected kernel modules (drivers).
> Then it will attempt to insert the patched module instead of the active one. If failed the original uvc modules will be restored.
> 
> ./scripts/patch-realsense-ubuntu-lts.sh

But my kernel is 4.20, so I got

`Unsupported kernel version 4.20.17 . The patches are maintained for Ubuntu16 (Xenial) with LTS kernel versions 4.4, 4.8, 4.10, 4.13 and 4.15
`

neilyoung, 2020-01-16:
The camera works perfectly on my mac:

2020-01-16 16:43:02,619 pose_pose.py INFO    : SDK: 2.32.0
2020-01-16 16:43:02,619 pose_pose.py INFO    : Firmware: 0.2.0.908
2020-01-16 16:43:02,619 pose_pose.py INFO    : Serial Number: 0000929122110112
2020-01-16 16:43:02,619 pose_pose.py INFO    : USB: 3.1

neilyoung, 2020-01-17:
I have completely reinstalled the system and compiled the dev branch again. NO DEVICE CONNECTED :(

I need to consider, that T265 and Banana PI is a no go.

But obviously I'm the only one who cares. Sad...

neilyoung, 2020-01-17:
To those who may be concerned:

On my freshly installed Banana PI M2 Zero, Ubuntu 16.04 with kernel 4.20.17 **the things are running fine, if I use the  librealsense `master` branch instead of `development`.**

I can only state, that there MUST still be an issue with USB with `development` branch in this  particular configuration.

I knew I had it running weeks ago already...

Do whatever you want with this info :)

neilyoung, 2020-01-17:
Hmmm. Wrong positive. Your USB handling is obviously not good at all.

```
13:50:42.115 [873] [E] Device-1400: MessageID 0x12 (DEV_START) failed with status 0x9
13:50:42.115 [873] [E] Device-1400: Error: Set Start Failed (0x1)

```

What a mess :/

neilyoung, 2020-01-17:
Giving up for now with the banana pi and return to PiZeroW. This thingy is not that powerfull, but at least it works on Raspbian Buster :(

neilyoung, 2020-01-19:
After a weekend of compilations (taking several hours each) here is my verdict:

Current DEVELOPMENT branch is absolutely NOT working with the BPi M2Zero, Ubuntu 16.04, Kernel 4.20.
Current MASTER branch is working sometimes.

It is a disastrous result of so much work and hope put into this device. And absolutely not understandable. Each f...g USB device of the world works with it, not so the Intel cameras.

My frustration is endless. Will return to the oversized RPIB3+ again, which is the ONLY one, which has kind of stable functionality with the T.265.
 
Closing this thread now. It makes no sense to bother.

closed 2020-01-19 by neilyoung

zainmehdi, 2020-01-20:
Hi @neilyoung I am not sure about version 2.32 but I was able to make it work with v 2.32.

neilyoung, 2020-01-20:
Somebody wrote me this morning. Can’t find his comment in the thread anymore; it was confusing either:

> Hi @neilyoung I am not sure about version 2.32 but I was able to make it work with v 2.32.

So if possible, unknown friend, please elaborate again. 

I for myself have not much to add: The latest development branch doesn’t work, master works partly. I yesterday gave up the M2 Banana Pi Zero in favor of a RPI 3B+ which works perfectly and is by far less bitchy (in terms of sensitivity against static charges and reliability in behavior at all)

zainmehdi, 2020-01-20:
@neilyoung I wrote it but I read that you already had been running it with previous versions so I deleted it. I too had been experimenting with many boards (Banana pi M2Z, Rpi3, RpiZW,RPi4, NanoPi Neo2, NanoPi Neo 2 Plus, NanoPi Neo4). The best of the bunch was NanoPi Neo4 with small form factor and high processing power. I was able to run T265 on all of these boards btw.

neilyoung, 2020-01-20:
@zainmehdi Thanks, yes in the past I was able to run it on RPiZeroW (no chance, too weak), Banana PI M2Z and RPI3B+. The Banana PI did work, doesn't work anymore. Are you using the development branch? If so, could you please communicate the commit you are using?

RPI3b+ is still working fine with the latest dev branch.

zainmehdi, 2020-01-21:
@neilyoung I used banana pi with v2.30 and v2.31. I read that you have been using Hardware reset lately. How can I get it ? V2.32 isnt available as of yet. Thanks

neilyoung, 2020-01-21:
@zainmehdi i‘m always just working with the development branch. There is the reset available. As said: My point is: 2.32 latest development branch is not working.

neilyoung, 2020-01-31:
closed for lack of interest