realsenseai/librealsense #5651: undefined reference to `rs2_get_error_message' when launching rs-multicam.cpp

karry3775, 2020-01-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  T265 | 
| Firmware Version                   | (None) | 
| Operating System & Version |   Linux (Ubuntu 118 LTS)  | 
| Kernel Version (Linux Only)    |  5.0.0-37-generic             | 
| Platform                                 | Laptop |
| SDK Version                            |  | 
| Language                            | C++| 
| Segment			|  Robot                          | 

### Issue Description
I have two t265 cameras which I want to extract pose information out of. I was trying to run rs-multicam.cpp file which upon execution of this command
g++ rs-multicam.cpp -o rs_main -lGL -lGLU -lglut

gives the attached error message. I wanted to know if there is away to fix this or any other work around that I could use to extract pose information from two t265s simulatneoulsy.

/tmp/cc2XtZLY.o: In function `rs2::error::error(rs2_error*)':
rs-multicam.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x2e): undefined reference to `rs2_get_error_message'
rs-multicam.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x73): undefined reference to `rs2_get_failed_function'
rs-multicam.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x9a): undefined reference to `rs2_get_failed_function'
rs-multicam.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0xfd): undefined reference to `rs2_get_failed_args'
rs-multicam.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x124): undefined reference to `rs2_get_failed_args'
rs-multicam.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x187): undefined reference to `rs2_get_librealsense_exception_type'
rs-multicam.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x19c): undefined reference to `rs2_free_error'
/tmp/cc2XtZLY.o: In function `rs2::error::handle(rs2_error*)':
rs-multicam.cpp:(.text._ZN3rs25error6handleEP9rs2_error[_ZN3rs25error6handleEP9rs2_error]+0x22): undefined reference to `rs2_get_librealsense_exception_type'
/tmp/cc2XtZLY.o: In function `rs2::stream_profile::stream_name[abi:cxx11]() const':
rs-multicam.cpp:(.text._ZNK3rs214stream_profile11stream_nameB5cxx11Ev[_ZNK3rs214stream_profile11stream_nameB5cxx11Ev]+0x5d): undefined reference to `rs2_stream_to_string'
/tmp/cc2XtZLY.o: In function `rs2::stream_profile::stream_profile(rs2_stream_profile const*)':
rs-multicam.cpp:(.text._ZN3rs214stream_profileC2EPK18rs2_stream_profile[_ZN3rs214stream_profileC5EPK18rs2_stream_profile]+0xc1): undefined reference to `rs2_get_stream_profile_data'
rs-multicam.cpp:(.text._ZN3rs214stream_profileC2EPK18rs2_stream_profile[_ZN3rs214stream_profileC5EPK18rs2_stream_profile]+0xe7): undefined reference to `rs2_is_stream_profile_default'
/tmp/cc2XtZLY.o: In function `rs2::frame::~frame()':
rs-multicam.cpp:(.text._ZN3rs25frameD2Ev[_ZN3rs25frameD5Ev]+0x23): undefined reference to `rs2_release_frame'
/tmp/cc2XtZLY.o: In function `rs2::frame::keep()':
rs-multicam.cpp:(.text._ZN3rs25frame4keepEv[_ZN3rs25frame4keepEv]+0x17): undefined reference to `rs2_keep_frame'
/tmp/cc2XtZLY.o: In function `rs2::frame::get_sensor()':
rs-multicam.cpp:(.text._ZN3rs25frame10get_sensorEv[_ZN3rs25frame10get_sensorEv]+0x35): undefined reference to `rs2_get_frame_sensor'
/tmp/cc2XtZLY.o: In function `rs2::frame::get_data() const':
rs-multicam.cpp:(.text._ZNK3rs25frame8get_dataEv[_ZNK3rs25frame8get_dataEv]+0x35): undefined reference to `rs2_get_frame_data'
/tmp/cc2XtZLY.o: In function `rs2::frame::get_profile() const':
rs-multicam.cpp:(.text._ZNK3rs25frame11get_profileEv[_ZNK3rs25frame11get_profileEv]+0x39): undefined reference to `rs2_get_frame_stream_profile'
/tmp/cc2XtZLY.o: In function `rs2::frame::add_ref() const':
rs-multicam.cpp:(.text._ZNK3rs25frame7add_refEv[_ZNK3rs25frame7add_refEv]+0x35): undefined reference to `rs2_frame_add_ref'
/tmp/cc2XtZLY.o: In function `rs2::frame::reset()':
rs-multicam.cpp:(.text._ZN3rs25frame5resetEv[_ZN3rs25frame5resetEv]+0x23): undefined reference to `rs2_release_frame'
/tmp/cc2XtZLY.o: In function `rs2::video_frame::video_frame(rs2::frame const&)':
rs-multicam.cpp:(.text._ZN3rs211video_frameC2ERKNS_5frameE[_ZN3rs211video_frameC5ERKNS_5frameE]+0x6d): undefined reference to `rs2_is_frame_extendable_to'
/tmp/cc2XtZLY.o: In function `rs2::video_frame::get_width() const':
rs-multicam.cpp:(.text._ZNK3rs211video_frame9get_widthEv[_ZNK3rs211video_frame9get_widthEv]+0x3d): undefined reference to `rs2_get_frame_width'
/tmp/cc2XtZLY.o: In function `rs2::video_frame::get_height() const':
rs-multicam.cpp:(.text._ZNK3rs211video_frame10get_heightEv[_ZNK3rs211video_frame10get_heightEv]+0x3d): undefined reference to `rs2_get_frame_height'
/tmp/cc2XtZLY.o: In function `rs2::points::get_vertices() const':
rs-multicam.cpp:(.text._ZNK3rs26points12get_verticesEv[_ZNK3rs26points12get_verticesEv]+0x3d): undefined reference to `rs2_get_frame_vertices'
/tmp/cc2XtZLY.o: In function `rs2::points::get_texture_coordinates() const':
rs-multicam.cpp:(.text._ZNK3rs26points23get_texture_coordinatesEv[_ZNK3rs26points23get_texture_coordinatesEv]+0x3d): undefined reference to `rs2_get_frame_texture_coordinates'
/tmp/cc2XtZLY.o: In function `rs2::motion_frame::motion_frame(rs2::frame const&)':
rs-multicam.cpp:(.text._ZN3rs212motion_frameC2ERKNS_5frameE[_ZN3rs212motion_frameC5ERKNS_5frameE]+0x6d): undefined reference to `rs2_is_frame_extendable_to'
/tmp/cc2XtZLY.o: In function `rs2::pose_frame::pose_frame(rs2::frame const&)':
rs-multicam.cpp:(.text._ZN3rs210pose_frameC2ERKNS_5frameE[_ZN3rs210pose_frameC5ERKNS_5frameE]+0x6d): undefined reference to `rs2_is_frame_extendable_to'
/tmp/cc2XtZLY.o: In function `rs2::pose_frame::get_pose_data() const':
rs-multicam.cpp:(.text._ZNK3rs210pose_frame13get_pose_dataEv[_ZNK3rs210pose_frame13get_pose_dataEv]+0x45): undefined reference to `rs2_pose_frame_get_pose_data'
/tmp/cc2XtZLY.o: In function `rs2::frameset::frameset(rs2::frame const&)':
rs-multicam.cpp:(.text._ZN3rs28framesetC2ERKNS_5frameE[_ZN3rs28framesetC5ERKNS_5frameE]+0x79): undefined reference to `rs2_is_frame_extendable_to'
rs-multicam.cpp:(.text._ZN3rs28framesetC2ERKNS_5frameE[_ZN3rs28framesetC5ERKNS_5frameE]+0xe2): undefined reference to `rs2_embedded_frames_count'
/tmp/cc2XtZLY.o: In function `rs2::frameset::operator[](unsigned long) const':
rs-multicam.cpp:(.text._ZNK3rs28framesetixEm[_ZNK3rs28framesetixEm]+0x67): undefined reference to `rs2_extract_frame'
/tmp/cc2XtZLY.o: In function `rs2::frame_queue::frame_queue(unsigned int, bool)':
rs-multicam.cpp:(.text._ZN3rs211frame_queueC2Ejb[_ZN3rs211frame_queueC5Ejb]+0x5b): undefined reference to `rs2_create_frame_queue'
rs-multicam.cpp:(.text._ZN3rs211frame_queueC2Ejb[_ZN3rs211frame_queueC5Ejb]+0x69): undefined reference to `rs2_delete_frame_queue'
[... 45 lines cut ...]
/tmp/cc2XtZLY.o: In function `window::window(int, int, char const*)::{lambda(GLFWwindow*, int, int, int, int)#4}::operator()(GLFWwindow*, int, int, int, int) const':
rs-multicam.cpp:(.text._ZZN6windowC4EiiPKcENKUlP10GLFWwindowiiiiE2_clES3_iiii[_ZZN6windowC4EiiPKcENKUlP10GLFWwindowiiiiE2_clES3_iiii]+0x26): undefined reference to `glfwGetWindowUserPointer'
/tmp/cc2XtZLY.o: In function `window::window(int, int, char const*)':
rs-multicam.cpp:(.text._ZN6windowC2EiiPKc[_ZN6windowC5EiiPKc]+0xde): undefined reference to `glfwInit'
rs-multicam.cpp:(.text._ZN6windowC2EiiPKc[_ZN6windowC5EiiPKc]+0xfa): undefined reference to `glfwCreateWindow'
rs-multicam.cpp:(.text._ZN6windowC2EiiPKc[_ZN6windowC5EiiPKc]+0x160): undefined reference to `glfwMakeContextCurrent'
rs-multicam.cpp:(.text._ZN6windowC2EiiPKc[_ZN6windowC5EiiPKc]+0x17a): undefined reference to `glfwSetWindowUserPointer'
rs-multicam.cpp:(.text._ZN6windowC2EiiPKc[_ZN6windowC5EiiPKc]+0x19f): undefined reference to `glfwSetMouseButtonCallback'
rs-multicam.cpp:(.text._ZN6windowC2EiiPKc[_ZN6windowC5EiiPKc]+0x1c4): undefined reference to `glfwSetScrollCallback'
rs-multicam.cpp:(.text._ZN6windowC2EiiPKc[_ZN6windowC5EiiPKc]+0x1e9): undefined reference to `glfwSetCursorPosCallback'
rs-multicam.cpp:(.text._ZN6windowC2EiiPKc[_ZN6windowC5EiiPKc]+0x20e): undefined reference to `glfwSetKeyCallback'
/tmp/cc2XtZLY.o: In function `window::~window()':
rs-multicam.cpp:(.text._ZN6windowD2Ev[_ZN6windowD5Ev]+0x1b): undefined reference to `glfwDestroyWindow'
[... cut ...]
rs-multicam.cpp:(.text._ZNK3rs211frame_queue14poll_for_frameINS_5frameEEENSt9enable_ifIXsrSt10is_base_ofIS2_T_E5valueEbE4typeEPS5_[_ZNK3rs211frame_queue14poll_for_frameINS_5frameEEENSt9enable_ifIXsrSt10is_base_ofIS2_T_E5valueEbE4typeEPS5_]+0x4e): undefined reference to `rs2_poll_for_frame'
/tmp/cc2XtZLY.o: In function `void rs2::processing_block::start<rs2::frame_queue>(rs2::frame_queue)':
rs-multicam.cpp:(.text._ZN3rs216processing_block5startINS_11frame_queueEEEvT_[_ZN3rs216processing_block5startINS_11frame_queueEEEvT_]+0x76): undefined reference to `rs2_start_processing'
collect2: error: ld returned 1 exit status

ev-mp, 2020-01-19:
@karry3775 hello , these are generic linkage errors due to the missing dependencies in the g++ invocation line -  `-lrealsense2` and other (lusb,...).
>g++ rs-multicam.cpp -o rs_main -lGL -lGLU -lglut

Also check whether #4398 can be useful.

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

Hi @karry3775

Were you able to test and try the suggestions from ev-mp?

karry3775, 2020-02-05:
Yes thank you! It works now

RealSenseCustomerSupport, 2020-02-06:
----------------------------------------------

That's great to hear.  I'll close the thread now.

closed 2020-02-06 by RealSenseCustomerSupport