realsenseai/librealsense #5652: How add the reference Intel.Realsense.dll in VB.net

Petter-zhang-NJ, 2020-01-15:
| Required Info
| Camera Model D435i
| Firmware Version 05.12.01.00
| Operating System & Version Win 10 Pro for OEM
| Platform PC
| SDK Version .NET 4.6
| Language VB.NET(2015)
| Segment Others
 I try to develop a app with VB.net for my D435i. I could already add the reference Intel.Realsense.dll, but  the realsense2.dll always failed. I do not know why.
 I found someone  have made it. 
Such as:
https://github.com/IntelRealSense/librealsense/issues/2019
How I add the reference  realsense2.dll in the VB.NET?

brettmilinski, 2020-01-23:
Try copying and pasting the realsense2.dll into your Debug folder manually using Windows File Explorer. I had to do this in C#. If you try to add the reference directly from Visual Studio it always fails. Make sure you copy the correct realsense2.dll in there though (x64 or x86) depending on what CPU configuration you are using.

Petter-zhang-NJ, 2020-02-01:
Thanks very much!
You are right. I added the reference Intel.Realsense.dll, and pasting the realsense2.dll into the Debug folder manually. It works with Vb.net.

closed 2020-02-01 by Petter-zhang-NJ