realsenseai/librealsense #5653: Record video stream simultaneously while using rs-trajectory

raghav123456, 2020-01-16:
Hello,
During working with the rs-trajectory example somehow i am able to save the pose data( Rotation and translation matrix ) which is generated by the example when moving the camera  now i need to save the video stream also of the  trajectory path simultaneously , so that i can record the pose data and video through the rs-trajectory example and later use for my project . 

So is their any way to record the video stream together when i move my t265 camera when run  
rs-trajectory .   
Thanks .

ev-mp, 2020-01-20:
@raghav123456  hello,
To include the video stream(s) along with the pose data you need to add these streams to `rs2::cfg` object that initializes that pipe and then retrieve and store the video frames on disk. Note that the trajectory itself is not a stream so you need to handle it in the code separately.
Also follow the #5671 thread with a similar request.

raghav123456, 2020-01-21:
@ev-mp Hello,thanks for your reply,
So what cfg we need to save both video and pose data , also what code i need to add or change so that video gets save.

raghav123456, 2020-01-23:
Friendly ping @ev-mp

closed 2020-10-30 by RealSenseSupport