realsenseai/librealsense #5654: RealSense error calling rs2_pipeline_start_with_config

Wintersunner, 2020-01-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hello
I was trying to rebuild some of the examples provided in the repo, first of all pretty much none of em can be build as they are. pretty much 99% the CMakelists file have to be modified to have realsense2 package so the project can be built.
I was trying the pose-estimation and pose examples I've noticed the error mentioned in the title.
Here's what I've done
I edited the CMakeLists.txt file to

```
# License: Apache 2.0. See LICENSE file in root directory.
# Copyright(c) 2019 Intel Corporation. All Rights Reserved.
#  minimum required cmake version: 3.1.0
cmake_minimum_required(VERSION 3.1.0)

project(RealsenseExamplesPose)
include(GNUInstallDirs)
find_package(realsense2 REQUIRED)
add_executable(rs-pose rs-pose.cpp)
set_property(TARGET rs-pose PROPERTY CXX_STANDARD 11)
target_link_libraries(rs-pose ${realsense2_LIBRARY} ${DEPENDENCIES})
set_target_properties (rs-pose PROPERTIES FOLDER Examples)
install(TARGETS rs-pose RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})
```
(The following is for the rs-pose example)
and built it in a directory named build inside example subdirectroy.
now when I try to run each of em I get the following error:

```
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55a981fb3980, config:0x55a981fac690):
    Couldn't resolve requests
```
Any help would be appreciated

zainmehdi, 2020-01-31:
I am having the same issue. It was working with version 2.31. @RealSenseCustomerSupport  please provide some insight here.

```
 //Turning off relocalization and Pose Jumping
    auto profile=cfg.resolve(pipe);
    dev=profile.get_device();
    sensor=dev.query_sensors().front();
    sensor.set_option(RS2_OPTION_ENABLE_RELOCALIZATION,0);
    sensor.set_option(RS2_OPTION_ENABLE_POSE_JUMPING,0);
    cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));

    auto PJ=sensor.get_option(RS2_OPTION_ENABLE_POSE_JUMPING);
    qDebug()<<"Pose Jumping ="<<PJ;

    auto RL=sensor.get_option(RS2_OPTION_ENABLE_RELOCALIZATION);
    qDebug()<<"Relocalization ="<<RL;

    qDebug()<<"Tracker Initialized";

    //Beacuse its the first run we dont want to reset tracker
    resetTracker=false;
    pipe.start(cfg);
```

RealSenseSupport, 2020-10-12:
Hi,

Are you still experiencing this issue on the latest release? We are able to build the SDK and run rs-pose on a T265 successfully on librealsense v2.38.1.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.38.1

Thanks

closed 2020-10-26 by RealSenseSupport