realsenseai/librealsense #5656: CMake Error at examples/hello-realsense/CMakeLists.txt:8 (add_executable):   Target "rs-hello-realsense" links to target "Threads::Threads" but the   target was not found.

jiangxkjohn, 2020-01-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  (4.15.0-74-generic                   | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
hello.everyone! I am a freshman in Realsense, when I followed the guide until the cmake process, I receive these cmake error. These warning and errors are all tell the links to "Threads::Threads", but I don't know what exactly it is. Do you know how to solve it? Thanks you!
`CMake Warning (dev) at src/gl/CMakeLists.txt:57 (add_library):
  Policy CMP0028 is not set: Double colon in target name means ALIAS or
  IMPORTED target.  Run "cmake --help-policy CMP0028" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target "realsense2-gl" links to target "Threads::Threads" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at examples/hello-realsense/CMakeLists.txt:8 (add_executable):
  Target "rs-hello-realsense" links to target "Threads::Threads" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at examples/software-device/CMakeLists.txt:12 (add_executable):
  Target "rs-software-device" links to target "Threads::Threads" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at examples/capture/CMakeLists.txt:9 (add_executable):
  Target "rs-capture" links to target "Threads::Threads" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?`

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

@jiangxkjohn Did you follow the guide in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md ?

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@jiangxkjohn Did you get it through? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2020-02-19:
----------------------------------------------

@jiangxkjohn Could you please update if you still need support on this? Thanks!

jiangxkjohn, 2020-02-19:
Sorry for my late reply, I have followed it but still not help, the error message is still "thread". I thought is was the case that there is no `set( CMAKE_CXX_FLAGS "-std=c++11" )`in CMakeLists.txt, but I am wrong.

jiangxkjohn, 2020-02-19:
@RealSenseCustomerSupport and I find that the first error in `CMake Error at examples/hello-realsense/CMakeLists.txt:8 (add_executable):`, there is no thread in that cpp file.

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

@jiangxkjohn Did you try the hint below provided by the error message?

 Run "cmake --help-policy CMP0028" for policy details.

jiangxkjohn, 2020-02-25:
sorry, I try using cmake_policy but it still cannot make any sense.

RealSenseCustomerSupport, 2020-03-03:
----------------------------------------------

Could you double check your cmake version? CMake 3.8+ is recommended.

"Threads::Threads" is a symbol in CMake.
https://cmake.org/cmake/help/v3.1/module/FindThreads.html

jiangxkjohn, 2020-03-03:
I have check my cmake version and it is 3.9.5, how strange it is.

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

There is another possibility. Is the build cache folder created by old version cmake?
BTW, have you install `libc6-dev` in your system? it will bring pthread.h to your system.
https://github.com/Kitware/CMake/blob/master/Modules/FindThreads.cmake#L128

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

@jiangxkjohn Did you get it through? Looking forward to your update. Thanks!

jiangxkjohn, 2020-03-10:
I have install libc6-dev and remove the cmake cache before rebuilding it again, but still cannot help

RealSenseCustomerSupport, 2020-03-12:
----------------------------------------------

@jiangxkjohn Sorry, I can't reproduce in my PC. Could you share more detail about your system?
I think CMake doesn't find the thread library(pthread in my PC).

jiangxkjohn, 2020-03-12:
Thank you for your help, I will find out the library and why, sorry, I am litte knowledge of Linux. If something goes well, I will inform you at once.

closed 2020-03-20 by RealSenseCustomerSupport