realsenseai/librealsense #5657: How to save bag files for T265 camera ???

Shivani1796, 2020-01-16:
Hi..
I have been using below code to save bag file for T265 camera, but the recorded bag file don't generate video .
```
rs2::config cfg;
cfg.enable_record_to_file("ab1.bag"); 
cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF); 
 pipe.start(cfg);
```

RealSenseCustomerSupport, 2020-01-24:
----------------------------------------------

Hi Shivani,

The behavior you are seeing is correct. If you enable a stream it disables all others. Here are some working examples that demonstrate how to record/playback bag files in librealsense:

https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback
https://github.com/IntelRealSense/librealsense/tree/master/tools/recorder

Thanks

RealSenseCustomerSupport, 2020-04-05:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

neilyoung, 2020-04-05:
You will have to enable both fisheye streams. That works, promised

closed 2020-04-26 by dorodnic