realsenseai/librealsense #5658: color_frame intrinsics vs align intrinsics (are same ?)

scizors, 2020-01-16:
D415 camera 
code :python

Problem: color intrinsics and align intrinsics are same why?
#5627 code color_intrinsics:
import pyrealsense2 as rs
import numpy as np
import cv2
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 6)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 6)
profile = pipeline.start(config)
frames = pipeline.wait_for_frames()
align_to = rs.stream.color
align = rs.align(align_to)
color_frame = frames.get_color_frame()
intrinsics=color_frame.profile.as_video_stream_profile().intrinsics
output color_intrinsics:
width: 1280, height: 720, ppx: 630.005, ppy: 376.119, fx: 925.769, fy: 925.476, model: 2, coeffs: [0, 0, 0, 0, 0]

#code:align intrinsics
import pyrealsense2 as rs
import numpy as np
import cv2
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 6)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 6)
profile = pipeline.start(config)
frames = pipeline.wait_for_frames()
align_to = rs.stream.color
align = rs.align(align_to)
aligned_frames = align.process(frames)
aligned_depth_frame = aligned_frames.get_depth_frame()
intrinsics = aligned_depth_frame.profile.as_video_stream_profile().intrinsics
output:align intrinsics
width: 1280, height: 720, ppx: 630.005, ppy: 376.119, fx: 925.769, fy: 925.476, model: 2, coeffs: [0, 0, 0, 0, 0]

ev-mp, 2020-01-19:
@scizors hello.
The intrinsic (as well as the extrinsic) parameters for the `aligned` stream are _borrowed_ from the stream the data is aligned to, i.e the color stream:
```python
align_to = rs.stream.color
align = rs.align(align_to)
```

scizors, 2020-01-20:
Hello,
 Thanks for the reply .
-I have used checker-board to calibrate the RGB camera D415 
-Then, I imported the images into Matlab finally I got the intrinsic values from Matlab
width: 1280, height: 720, ppx: 628.505325548172 , ppy: 375.139062149491 , fx: 908.237035106199 , fy: 908.710269684772 

From intelrealsense 
width: 1280, height: 720, ppx: 630.005, ppy: 376.119, fx: 925.769, fy: 925.476, model: 2, coeffs: [0, 0, 0, 0, 0]
camera is already well calibrated while buying it. why there is major difference in the focal length (fx and fy)? will this deviation affect the accuracy ? 

which parameters should I use now ?

ev-mp, 2020-01-20:
@scizors, 
The focal length affects the object's magnification, so in case fx/fy are skewed the size of the objects would be scaled and it is quite straight-forward to devise a test that will measure ruler's size for verification.
A more methodical approach is to use the [DepthQuality white-paper](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_DepthQualityTesting.pdf) and the accompanying `rs-depth-quality` tool to characterize depth data and to see whether the metrics are within the specifications.

As for using 3rd part calibration with camera - currently this option is relevant only for OEM customers. 
When it come to the usage then with the live camera you can only use the factory calibration (that has been [recently extended](https://github.com/IntelRealSense/librealsense/pull/5346) with on-chip calibration feature)
The way you can try the alternative calibration is wrap the live device with `software_device` and have it calibration established with Matlab-produced data.

ev-mp, 2020-02-03:
Assume resolved

closed 2020-02-03 by ev-mp