realsenseai/librealsense #5660: some questions about using OpenCV to save and show depth image

qiyueliao, 2020-01-16:
My camera is D435i, and I use Opencv 3.4.7 、realsense SDK 2.0 in Ubuntu 18.04. After I align the color and depth stream ,I use Mat depth_image to save the depth data of the depth frame ,and  use imwrite() to save Mat depth_image to Depth.png, then use imshow() to show the Mat .

My questions are:

1、why the Depth.png looks different  from the imshow window? They both come from Mat depth_image.
2、Can I think the data of the pixel in Depth.png is the depth information I get from the camera's depth frame? Is the unit  millimeter? 
I would be grateful for your help!

main.cpp

#include <iostream>
#include <sstream>
#include <fstream>
#include <algorithm>
#include <cstring>

#include<opencv2/imgproc/imgproc.hpp>
#include<opencv2/core/core.hpp>
#include<opencv2/highgui/highgui.hpp>

#include<librealsense2/rs.hpp>
using namespace std;
using namespace cv;

int main(int argc, char * argv[])
{
rs2::pipeline pipe;
rs2::config pipe_config;
pipe_config.enable_stream(RS2_STREAM_DEPTH,640,480,RS2_FORMAT_Z16,30);
pipe_config.enable_stream(RS2_STREAM_COLOR,640,480,RS2_FORMAT_BGR8,30);
rs2::pipeline_profile profile = pipe.start(pipe_config);

rs2::frameset frameset = pipe.wait_for_frames();

rs2_stream align_to = RS2_STREAM_COLOR;
rs2::align align(align_to);
auto processed = align.process(frameset);

rs2::depth_frame depth = processed.get_depth_frame();

const int depth_w=depth.as<rs2::video_frame>().get_width();
const int depth_h=depth.as<rs2::video_frame>().get_height();

Mat depth_image(Size(depth_w,depth_h),CV_16SC1,(void*)depth.get_data(),Mat::AUTO_STEP);

imwrite("Depth.png",depth_image);
imshow("depth used imshow",depth_image);

waitKey(0);
return 0;
}

CMakeLists.txt

cmake_minimum_required(VERSION 2.8 FATAL_ERROR)
project(Align)

find_package(OpenCV REQUIRED)

include_directories(${OpenCV_INCLUDE_DIRS})

set(DEPENDCIES realsense2)

add_executable(Align main.cpp)
target_link_libraries(Align ${OpenCV_LIBS} ${DEPENDCIES})
Depth.png

what I see in imshow window:
(https://user-images.githubusercontent.com/42409750/72529521-7235a180-38a8-11ea-8348-c96270c133c5.png)
what I see in imshow window:
(https://user-images.githubusercontent.com/42409750/72529646-add06b80-38a8-11ea-90df-2ba54b2cedb9.png)

kafan1986, 2020-01-17:
> My camera is D435i, and I use Opencv 3.4.7 、realsense SDK 2.0 in Ubuntu 18.04. After I align the color and depth stream ,I use Mat depth_image to save the depth data of the depth frame ,and use imwrite() to save Mat depth_image to Depth.png, then use imshow() to show the Mat .
> 
> My questions are:
> 
> 1、why the Depth.png looks different from the imshow window? They both come from Mat depth_image.
> 2、Can I think the data of the pixel in Depth.png is the depth information I get from the camera's depth frame? Is the unit millimeter?
> I would be grateful for your help!
> 
> main.cpp
> 
> #include
> #include
> #include
> #include
> #include
> 
> #include<opencv2/imgproc/imgproc.hpp>
> #include<opencv2/core/core.hpp>
> #include<opencv2/highgui/highgui.hpp>
> 
> #include<librealsense2/rs.hpp>
> using namespace std;
> using namespace cv;
> 
> int main(int argc, char * argv[])
> {
> rs2::pipeline pipe;
> rs2::config pipe_config;
> pipe_config.enable_stream(RS2_STREAM_DEPTH,640,480,RS2_FORMAT_Z16,30);
> pipe_config.enable_stream(RS2_STREAM_COLOR,640,480,RS2_FORMAT_BGR8,30);
> rs2::pipeline_profile profile = pipe.start(pipe_config);
> 
> rs2::frameset frameset = pipe.wait_for_frames();
> 
> rs2_stream align_to = RS2_STREAM_COLOR;
> rs2::align align(align_to);
> auto processed = align.process(frameset);
> 
> rs2::depth_frame depth = processed.get_depth_frame();
> 
> const int depth_w=depth.asrs2::video_frame().get_width();
> const int depth_h=depth.asrs2::video_frame().get_height();
> 
> Mat depth_image(Size(depth_w,depth_h),CV_16SC1,(void*)depth.get_data(),Mat::AUTO_STEP);
> 
> imwrite("Depth.png",depth_image);
> imshow("depth used imshow",depth_image);
> 
> waitKey(0);
> return 0;
> }
> 
> CMakeLists.txt
> 
> cmake_minimum_required(VERSION 2.8 FATAL_ERROR)
> project(Align)
> 
> find_package(OpenCV REQUIRED)
> 
> include_directories(${OpenCV_INCLUDE_DIRS})
> 
> set(DEPENDCIES realsense2)
> 
> add_executable(Align main.cpp)
> target_link_libraries(Align ${OpenCV_LIBS} ${DEPENDCIES})
> 
> what I see in imshow window:
> (https://user-images.githubusercontent.com/42409750/72529521-7235a180-38a8-11ea-8348-c96270c133c5.png)
> what I see in imshow window:
> (https://user-images.githubusercontent.com/42409750/72529646-add06b80-38a8-11ea-90df-2ba54b2cedb9.png)

Raw depth values can be from 0 to 65535, so Mat type needs to be CV_16UC1

`Mat depth_image(Size(depth_w,depth_h),CV_16UC1,(void*)depth.get_data(),Mat::AUTO_STEP);`

For visualization (in gray form) it is better to convert to CV_8UC1.

`depth_image.convertTo(depth_image, CV_8UC1, 1/255.0);`

qiyueliao, 2020-01-17:
Thank you ,the first problem is solved. what about the sencond? Thank you for your reply!

kafan1986, 2020-01-17:
> Thank you ,the first problem is solved. what about the sencond? Thank you for your reply!

Yes the CV_16UC1 Mat contains depth value in millimeter. You can access the depth value directly from Mat. If you have used convertTo function then you have divided the depth value by 255.0, so if you are using this png to load on to a Mat later to be used, make sure you calculate using that factor.

qiyueliao, 2020-01-17:
sorry, I have another question. As the raw depth values are from 0 to 65535, the pixel value of the CV_16UC1 Mat that get the depth value are aslo from 0 to 65535 ,but when I insert code below to output the pixel value , some seem larger than 65535.
int n=0;
for (int v = 0; v < depth_image.rows; v++)
        {
                for (int u = 0; u < depth_image.cols; u++) {
              
                unsigned int d=depth_image.at<unsigned int>(v,u);// read the depth value
                        if (d == 0) continue; // if depth value is zero , continue

                if(n<50){std::cout<<d<<std::endl;n++;}//output 50 depth value
                }
        }
the output:
(https://user-images.githubusercontent.com/42409750/72610148-9d2ffc00-3961-11ea-8de0-e3fb5f19f2f4.png)

kafan1986, 2020-01-17:
Your problem are opencv related. I have not used c++ with opencv much. You are using wrong data type for given Mat type. Anways use ushort instead of unsigned int.

qiyueliao, 2020-01-17:
Thank you very much ! After change the data type, I think I get the right value. Thank you !

closed 2020-01-24 by RealSenseCustomerSupport