realsenseai/librealsense #5661: A problem when open the bag file

princetonpku, 2020-01-16:
1、I save the bag file frame by frame by the following function：
```
void saveSingleFrame(const rs2::frame &data, const std::string &fileFolder)
{
	rs2::software_device dev;
	std::vector<std::tuple<rs2::software_sensor, rs2::stream_profile, int>> sensors;
	std::vector<std::tuple<rs2::stream_profile, rs2::stream_profile>> extrinsics;
       if (auto fs = data.as<rs2::frameset>()) 
	{
		for (int i = 0; i < fs.size(); ++i) 
		{
			rs2::frame f = fs[i];
			auto profile = f.get_profile();
			std::stringstream sname;
			sname << "Sensor (" << i << ")";
			rs2::software_sensor s = dev.add_sensor(sname.str());
			rs2::stream_profile software_profile;

			if (auto vf = f.as<rs2::video_frame>()) 
			{
				auto vp = profile.as<rs2::video_stream_profile>();
				rs2_video_stream stream{ vp.stream_type(), vp.stream_index(), i, vp.width(), vp.height(), vp.fps(), vf.get_bytes_per_pixel(), vp.format(), vp.get_intrinsics() };
				software_profile = s.add_video_stream(stream);
				if (f.is<rs2::depth_frame>()) {
					auto ds = sensor_from_frame(f)->as<rs2::depth_sensor>();
					s.add_read_only_option(RS2_OPTION_DEPTH_UNITS, ds.get_option(RS2_OPTION_DEPTH_UNITS));
				}
			}
			else if (f.is<rs2::motion_frame>()) 
			{
				auto mp = profile.as<rs2::motion_stream_profile>();
				rs2_motion_stream stream{ mp.stream_type(), mp.stream_index(), i, mp.fps(), mp.format(), mp.get_motion_intrinsics() };
				software_profile = s.add_motion_stream(stream);
			}
			else if (f.is<rs2::pose_frame>()) {
				rs2_pose_stream stream{ profile.stream_type(), profile.stream_index(), i, profile.fps(), profile.format() };
				software_profile = s.add_pose_stream(stream);
			}
			else {
				// TODO: How to handle other frame types? (e.g. points)
				assert(false);
			}
			sensors.emplace_back(s, software_profile, i);

			bool found_extrin = false;
			for (auto& root : extrinsics) {
				try {
					std::get<0>(root).register_extrinsics_to(software_profile,
						std::get<1>(root).get_extrinsics_to(profile)
					);
					found_extrin = true;
					break;
				}
				catch (...) {}
			}
			if (!found_extrin) {
				extrinsics.emplace_back(software_profile, profile);
			}
		}
		// Recorder needs sensors to already exist when its created
		std::stringstream name;
		name << fileFolder<<std::setw(5)<<std::setfill('0')<<data.get_frame_number() << ".bag";
		rs2::recorder rec(name.str(), dev, false);

		for (auto group : sensors) 
		{
			auto s = std::get<0>(group);
			auto profile = std::get<1>(group);
			s.open(profile);
			s.start([](rs2::frame) {});
			rs2::frame f = fs[std::get<2>(group)];
			if (auto vf = f.as<rs2::video_frame>()) {
				s.on_video_frame({ const_cast<void*>(vf.get_data()), [](void*) {}, vf.get_stride_in_bytes(), vf.get_bytes_per_pixel(),
								   vf.get_timestamp(), vf.get_frame_timestamp_domain(), static_cast<int>(vf.get_frame_number()), profile });
			}
			else if (f.is<rs2::motion_frame>()) {
				s.on_motion_frame({ const_cast<void*>(f.get_data()), [](void*) {}, f.get_timestamp(),
									f.get_frame_timestamp_domain(), static_cast<int>(f.get_frame_number()), profile });
			}
			else if (f.is<rs2::pose_frame>()) {
				s.on_pose_frame({ const_cast<void*>(f.get_data()), [](void*) {}, f.get_timestamp(),
								  f.get_frame_timestamp_domain(), static_cast<int>(f.get_frame_number()), profile });
			}
			s.stop();
			s.close();
		}
	};

};
```
I can open the bag file by the realsense-viewer.exe。However I cannot get the color frame by the following：
```
std::shared_ptr<rs2::pipeline> pipe = std::make_shared<rs2::pipeline>();
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR);
cfg.enable_stream(RS2_STREAM_DEPTH);
cfg.enable_device_from_file(“0001.bag”);
pipe->start(cfg);
auto frameSet = pipe->wait_for_frames();
rs2::frame depth = frameSet.get_depth_frame();
rs2::frame color = frameSet.get_color_frame();
```
I think there are some difference on open bag file. So I try using the following code according the source code from realsense-viewer.exe. But it doesn't work.

```
rs2::context ctx;
rs2::playback dev = ctx.load_device("00001.bag");
std::shared_ptr<syncer_model> syncer = std::make_shared<syncer_model>();;
syncer->start();
std::shared_ptr<rs2::asynchronous_syncer> dev_syncer = syncer->create_syncer();
std::map<int, rs2::frame_queue> frames_queue;
for (auto sub : dev.query_sensors())
{
    std::vector<rs2::stream_profile> sensor_profiles = sub.get_stream_profiles();;
    sub.open(sensor_profiles.front());
    sub.set_option();
    //sub.start(sync);
    sub.start([&, dev_syncer](rs2::frame f)
    {
        //dev_syncer->invoke(f);
        auto id = f.get_profile().unique_id();
        frames_queue[id].enqueue(f);
        on_frame();
    });
}
```
Could someone help me? and explan the framework of realsense-viewer.exe when open one bag file.
Thank you!

princetonpku, 2020-01-16:
@Sergey Dorodnicov

RealSenseCustomerSupport, 2020-05-03:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-05-11 by RealSenseCustomerSupport