realsenseai/librealsense #5662: fails to install on linux kernel /usr/src/linux-headers-5.4.0-9-generic

scottstensland, 2020-01-16:
fails to install on linux kernel 5.x

    x86_64
    5.4.0-9-generic
    DISTRIB_ID=Ubuntu

    uname -a
    Linux peach 5.4.0-9-generic #12-Ubuntu SMP Mon Dec 16 22:34:19 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

sudo apt-get install librealsense2-dkms

<details>
<summary>
here is output of above install with errors
Building for 5.4.0-9-generic
Building initial module for 5.4.0-9-generic
Error! Bad return status for module build on kernel: 5.4.0-9-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.9/build/make.log for more information.
dpkg: error processing package librealsense2-dkms (--configure):

</summary>

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done

The following additional packages will be installed:
  librealsense2-udev-rules
The following NEW packages will be installed:
  librealsense2-dkms librealsense2-udev-rules
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 7,832 kB of archives.
After this operation, 61.8 MB of additional disk space will be used.
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 librealsense2-dkms all 1.3.9-0ubuntu1 [7,825 kB]
0% [1 librealsense2-dkms 14.3 kB/7,825 kB 0%]
Get:2 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 librealsense2-udev-rules amd64 2.31.0-0~realsense0.1790 [6,794 B]
Fetched 7,832 kB in 3s (2,989 kB/s)            
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 312638 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.9-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.9-0ubuntu1) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../librealsense2-udev-rules_2.31.0-0~realsense0.1790_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.31.0-0~realsense0.1790) ...
Setting up librealsense2-udev-rules:amd64 (2.31.0-0~realsense0.1790) ...
Setting up librealsense2-dkms (1.3.9-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.9 DKMS files...
Building for 5.4.0-9-generic
Building initial module for 5.4.0-9-generic
Error! Bad return status for module build on kernel: 5.4.0-9-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.9/build/make.log for more information.
dpkg: error processing package librealsense2-dkms (--configure):
 installed librealsense2-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 librealsense2-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

```
</details>

here is your crash log file

	cat /var/lib/dkms/librealsense2-dkms/1.3.9/build/make.log 
	DKMS make.log for librealsense2-dkms-1.3.9 for kernel 5.4.0-9-generic (x86_64)
	Thu 16 Jan 2020 12:15:46 PM EST
	make: Entering directory '/usr/src/linux-headers-5.4.0-9-generic'
	arch/x86/Makefile:147: CONFIG_X86_X32 enabled but no binutils support
	scripts/Makefile.build:42: /var/lib/dkms/librealsense2-dkms/1.3.9/build/5.4.0/drivers/media/usb/uvc/Makefile: No such file or directory
	make[1]: *** No rule to make target '/var/lib/dkms/librealsense2-dkms/1.3.9/build/5.4.0/drivers/media/usb/uvc/Makefile'.  Stop.
	make: *** [Makefile:1681: /var/lib/dkms/librealsense2-dkms/1.3.9/build/5.4.0/drivers/media/usb/uvc] Error 2
	make: Leaving directory '/usr/src/linux-headers-5.4.0-9-generic'

ev-mp, 2020-01-19:
@scottstensland hello,
There are no dkms/patches for kernel 5.4. The closest one is LTS kernel 5.3 support for which is scheduled in February.
From the supported LTS kernels the closest is 5.0.0-xx. Installing patches for 5.0.0-xx is available with [by running the script](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)  
```sh
./scripts/patch-realsense-ubuntu-lts.sh
```

The current LTS support status is maintained on the [installation page](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution). As a side note in the next DKMS update we'll ensure to alarm kernel version (non)compatibility to avoid inconvenience.

mikepurvis, 2020-02-27:
Ubuntu Focal LTS comes out in April and has 5.4 as the default kernel: https://packages.ubuntu.com/source/focal/linux

We're beginning to test with this OS and would appreciate support for it. Additionally, it would be great to have the source for the dkms package opened up so that it's straightforward for third parties to modify/update the patches as required and push their own versions of the package to PPAs.

achirkin, 2020-04-11:
Focal fossa beta is out now. I've just switched to it and join the request.

This morning, I tried to build everything from source. It turned out to patch and compile well with very small changes from 5.3. [Here is my workaround](https://github.com/achirkin/librealsense/commit/74d0101ee8199dab955f2d59b97c1734c5995198). I successfully ran realsense-viewer and the grabcuts example (with opencv 4 from ubuntu ppa). Works for me until the official support is out! (also note I am using lowlatency kernel instead of generic).

breznak, 2020-04-23:
New Ubuntu LTS is released today, can someone elaborate when we can expect it supported by RS? 
https://releases.ubuntu.com/20.04/

ruffsl, 2020-05-01:
> Additionally, it would be great to have the source for the dkms package opened up so that it's straightforward for third parties to modify/update the patches as required and push their own versions of the package to PPAs.

That would be awesome with ROS1 Noetic and ROS2 Foxy coming out soon!

ruffsl, 2020-05-01:
> It turned out to patch and compile well with very small changes from 5.3. Here is my workaround.

@achirkin , could open a PR here with your branch to get the ubuntu focal ball rolling?

chrisgraf, 2020-09-25:
Is there a timeframe when ubuntu 20.04 LTS & kernel 5.4 will be supported?

pedrombmachado, 2020-09-27:
We really need the librealsense2 for Ubuntu 20.04 because it is stable and it is the new LTS. What is the plan?

slitcch, 2020-10-28:
When is this gonna happen?

breznak, 2020-10-28:
Guys, we figured there are somehow 2 "official" repos for RS, and switched to this one which is much better supported for ROS2, recent Ubuntu etc. 
https://github.com/intel/ros2_intel_realsense

pedrombmachado, 2020-11-03:
@dorodnic do you know anything about when the librealsense will be available for Ubuntu 20.04 LTS?

MartyG-RealSense, 2020-11-24:
Hi everyone,  

librealsense SDK 2.40.0 now officially supports Kernel 5,4 and has alpha-level support for Ubuntu 20 (Focal).

https://github.com/IntelRealSense/librealsense/releases/tag/v2.40.0

RealSenseSupport, 2020-12-05:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

pedrombmachado, 2020-12-05:
Working for me on Ubuntu 20.04 with kernel 5.4.0-54-generic.

closed 2020-12-14 by RealSenseSupport