realsenseai/librealsense #5666: Director

DavidCadtec, 2020-01-17:
I own a D415
Can anyone recommend any lbrary of bag files ?
More specifically I neef a repository of bag files of pedestrian circulation indoor records.

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @DavidCadtec,

You can find a couple of sample files here:
https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

Thank you!

RealSenseCustomerSupport, 2020-02-01:
----------------------------------------------

@DavidCadtec,

Do you need any further assistance with this question?

DavidCadtec, 2020-02-01:
Thanks for your recommendation ...
The resources you listes don't have a people track example...
If there's not other resource for bag files i will try to record some bag files myself...
I tried however to use the SDK for that but after recording the bag file could not be opened in the RealSense Viewer (Bag file not indexed error...).
Any hints to get that fixed ?

Thanks !

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

@DavidCadtec,

Can you please provide more information about the problem with bag files recording/playback?
What is exact scenario of the issue?

Thank you!

DavidCadtec, 2020-02-05:
Sure.
I'm recording the bag file via Pyrealsense2 SDK (Python Wrapper)... the code i'm using is as below ...
After run it i get the ros bag file saved in the folder ...
But when i try to open it in the RealSense Viewer i get::

"Failed to load file /home/davidpro/PycharmProjects/EstudosIntelRS/test.bag. Reason: Io in rs2_context_add_device(ctx:0x5cd1890, file:/):
Failed to create ros reader: Bag unindexed"

I'm running the code in the PyCharm IDE
Ubuntu 16.04
Follows the code i'm using:

 # Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

config.enable_record_to_file("test.bag")
#rs.recorder("test2.bag")

# Start streaming
pipeline.start(config)

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        # Stack both images horizontally
        images = np.hstack((color_image, depth_colormap))

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)

        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            [break](url)
finally:
    # Stop streaming
    pipeline.stop()

RealSenseCustomerSupport, 2020-02-10:
----------------------------------------------

@DavidCadtec,

Do you experience the same issue when you are recording using LRS without Python Wrapper?

Thank you!

RealSenseCustomerSupport, 2020-02-19:
----------------------------------------------

@DavidCadtec,

Did you have a chance to record .bag files using LRS without Python Wrapper?

DavidCadtec, 2020-02-22:
LRS stands for Lib RealSense correct ? No, i did not ... i'm not fluent at all in C++ ... all i can use is Python. But i have seen some posts of other people facing the same issue also in SDK

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

@DavidCadtec,

Please try to record a .bag file using the latest version of RealSense Viewer and let us
know if it works for you.

closed 2020-03-09 by RealSenseCustomerSupport