realsenseai/librealsense #5667: C++ not reading depth data from .bag file

SB11416158, 2020-01-17:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            | 2                         | 
| Language                            |  C++ / Python                          | 

Issue Description:

I am using a D435 camera to record a .bag file and later using this recorded file to create point clouds. When I am reading in the .bag in C++ all the depth values are 0. However if I read in the same .bag in Pyhton it has all the correct depth values. I need to use C++ but I can't get the depth values from the file. Below are short copies of both codes:

----------------------------------------------------------------------------------------------------------
C++:
```
	rs2::pipeline pipeline;
	rs2::config cfg;
	cfg.enable_device_from_file(BagFilePath, false);

	pipeline.start(cfg);

	rs2::frameset frames = pipeline.wait_for_frames(); 
	rs2::depth_frame depth_frame = frames.get_depth_frame();	

	const int width = depth_frame.get_width();
	const int height = depth_frame.get_height();

	std::cout << "Distance at centre is " << depth_frame.get_distance(int(width / 2), int(height / 2)) << " meters.";
	
	pipeline.stop();
```
C++ result: "Distance at centre is 0 meters."

----------------------------------------------------------------------------------------------------------
Python:
```
pipeline = rs.pipeline()
cfg = rs.config()

rs.config.enable_device_from_file(cfg, BagFilePath, False)

pipeline.start(cfg)

frames = pipeline.wait_for_frames()
depth_frame = frames.get_depth_frame()

width = depth_frame.get_width()
height = depth_frame.get_height()

print("Distance at centre is  ", depth_frame.get_distance(int(width/2), int(height/2)), " meters.")

pipeline.stop()  
```
Python result: "Distance at centre is   0.45600003004074097  meters.."

ev-mp, 2020-01-19:
@SB11416158 hello,
The depth fill-rate is generally less than 100%, and having zero-data pixels in each frame is normal.
The code snippets suggest that you sample only one frame. Have you tried to check the distance for 20-30 consecutive frames, or look at the the neighboring pixels for w/2, h/2 ?

Additionally, when running non real-time playback you can correlate the depth data obtained with the frame index and check whether the results obtained with python and c++ are consistent.

SB11416158, 2020-01-20:
Hi @ev-mp ,

Thank you for your reply and suggestions. You're right I am only trying on one frame. Over the whole image all depth values on C++ are 0 but in Python I get depth values. I have tried over several consecutive frames with the same result. C++ always gives a 0 depth value while Python gives proper depth values.

ev-mp, 2020-01-20:
@SB11416158 ,  The python code is a wrapper for the functions you call in C++ code, so the definition can't explain the results. On the contrary - in case of discrepancy it is more likely that the C++data becomes corrupted/zeroed when converted to Python but not the other way around.

The first thing to check is whether you get the same depth frame continuously in
>rs2::frameset frames = pipeline.wait_for_frames(); 
rs2::depth_frame depth_frame = frames.get_depth_frame();

Print out the frame attributes, such as timestamp and frame number for verification. Also - please check that the realtime flag value is identical.

Additionally - try to replay the file using `realsense-viewer` and report the results.

SB11416158, 2020-01-20:
Hi @ev-mp , thanks for your response. I checked the time stamp and frame number and they are changing. I compared the timestamp and frame number for the same .bag file in C++ and Python and for the same frame number they have different timestamps.
Running the same .bag file in realsense-viewer all the depth values are 0.
I'm not sure how to check the realtime flag values. How do I check this?

I downloaded the stairs.bag from realsense and ran this. All C++, Python and the realsense viewer are able to read the depth values from the stairs.bag file.

Perhaps I am making an error in the way I am saving my .bag files?

ev-mp, 2020-01-21:
@SB11416158 , can you share the .bag for review?

Also put a nullptr check for the result of `rs2::depth_frame depth_frame = frames.get_depth_frame();`  both in c++ and python.

SB11416158, 2020-01-21:
Hi @ev-mp , I added the nullptr check and neither are showing an error, so both C++ and Python are getting the depth frame.
Sure, I have shared the files with you

ev-mp, 2020-01-21:
@SB11416158 thank you for the references provided.
I just finished downloading the data and will look into it.

SB11416158, 2020-01-30:
Hi @ev-mp ,
Any update on this? Were you able to replicate the problem?

ev-mp, 2020-02-03:
@SB11416158 , sorry for the delay- I checked both the bag files provided several days ago and they were both correct. I managed to replay both the single frame and the longer records.

Just to confirm - did you read the zero values for the central pixel in both bag files, or only the shorter one? 
I think that what's going on is that after starting the playback the recorded frames are being replayed faster than you manage to poll it. This could be resolved when you turn the `realtime playback` flag to false. 
```python
config.enable_device_from_file('/plant.bag',repeat_playback=False)
profile = pipe.start(config)
playback = profile.get_device().as_playback()
playback.set_real_time(False)
```
see #5107. 

Additionally you can verify it by setting the `replay` flag to true, so that even in a case of  a single frame it will be send multiple times allong for `pipe.wait_for_frames()` to succeed.
>cfg.enable_device_from_file(BagFilePath, ~~false~~**true**);

SB11416158, 2020-02-04:
Hi @ev-mp , 
Thanks for you reply.

To confirm, I have recorded .bag files using a Python script.
When I playback these .bag files in Python, everything is ok and I have all the depth values for all pixels except for some holes.
But when I playback these .bag files in C++ all depth values are zero for all pixels. Also playing back these .bag files in RealSense Viewer all depth values are zero for all pixels.
I need to get the depth values from the .bag files in C++ because I am working with the PCL library in C++ to work with point clouds.

I have tried your suggestions of setting the realtime playback flag to false but still in C++ I have zero depth for all pixels

RealSenseCustomerSupport, 2020-05-03:
----------------------------------------------

Hi.

Will you be needing further help with this? The fact that we can successfully run the bag files you provided suggests this is not a librealsense issue. If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-05-11 by RealSenseCustomerSupport