realsenseai/librealsense #5668: Cameras disappearing during runtime or at boot

SFR-mir, 2020-01-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.6.250 | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-38-generic                                         | 
| Platform                                 | DFI  |
| SDK Version                            |  2.25                           | 
| Language                            |  C++                           | 
| Segment			|  Robot                           | 

### Issue Description

We have a robot with 2 Intel Realsense cameras connected to a DFI industrial computer. 
We have observed during multiple incidents that either one or more cameras will disappear seamingly at random and it is not able to recover even after executing recovery behaviours such as hardware_reset https://github.com/IntelRealSense/librealsense/issues/1086 through the SDK.
When this happens, we are not able to query the cameras through *rs-enumerate-devices*.
The only solution that is reliable is powering down the system completely for an extended time (30+ minutes) or most of the times it works with physically replugging the cameras. None of these is not a valid solution for us.
By reading associated issues, we came across solutions that suggest bumping the firmware to the latest recommended version (we did that recently) https://github.com/IntelRealSense/librealsense/issues/4457,
solutions that suggest changing the USB cables (we have tested many, the provided ones and high end aftermarket but the issue persists).
We also tested the suggestion from https://github.com/IntelRealSense/librealsense/issues/4552#issuecomment-518370718 to bind/unbind the xhci_hd driver but it does not help in our case.
Finally we have performed the test from https://github.com/NoMagicAi/reproduce_d400_camera_issue. We perform the test 100 times with 4 D435 cameras attached to a single DFI which resulted in 2 cameras disappearing after a short while and they didn't recover even through shutting down the machine for 3 minutes. 
I've attached the output of the before-mentioned test in *experiment.log* and the output of dmesg after each experiment in *syslog.log*.
We also caught one of the random camera disconnects/reconnect after which it didn't work and we performed a hardware reset and it still didn't work. I've attached the syslog from that event and called it  *camera_randomly_disappearing.log*
[syslog.log](https://github.com/IntelRealSense/librealsense/files/4077126/syslog.log)
[realsense.log](https://github.com/IntelRealSense/librealsense/files/4077127/realsense.log)
[experiment.log](https://github.com/IntelRealSense/librealsense/files/4077128/experiment.log)
[camera_randomly_disappearing.log](https://github.com/IntelRealSense/librealsense/files/4077130/camera_randomly_disappearing.log)

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

Hi SFR-mir,
(1) the SDK API you used is old, (v2.25)
we have new version SDK that is version 2.32 and newer FW which is 5,12.2
would suggest you to use that latest SDK (API) and FW  that improve the reliability of streaming

(2) for multi-camera streaming, we have white paper to ensure you configure the camera properly.
https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration
there are something you need to pay attention to such as power and bandwidth..

if you still have issue after using new SDK and FW
please see if reduce the resolution or unplug some camera can reduce the frequency of occurrence

In general the power consumption can approach or even exceed 2W per camera. As such, a rule of thumb should be to make sure to use USB Hubs that support external power supplies, or to use individual ports on a PC that provide independent power to the USB spec. A powered HUB will normally provide 12.5W or more.

as to the USB cable, the length matter which also mentioned at issue#4452
https://github.com/IntelRealSense/librealsense/issues/4552
not just affect the signal quality but also the VBUS power level,  the longer cable, the lower VBUS voltage level

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Hi @SFR-mir,
Could  provided suggestion can help on your issue?

RealSenseCustomerSupport, 2020-03-02:
----------------------------------------------

Hi @SFR-mir

besides checking
 the power consumption or cable quality

We have a newer viewer 2.33.1 and FW just released with below fix

**FW 5.12.3.0 for D4xx Devices** Stability and Power Management fixes and enhancements

Please see if this can fix the problem you have now

RealSenseCustomerSupport, 2020-03-09:
----------------------------------------------

Hi @SFR-mir,
is this still an issue for you
or the suggestion provided can help to mitigate your issue?

SFR-mir, 2020-03-10:
Hi @RealSenseCustomerSupport,
I'll update the firmware and see if it solves the issue.

RealSenseCustomerSupport, 2020-04-06:
----------------------------------------------

Hi SFR-mir,
could FW update solve your issue already?
and please validate it when only plug one camera to the host first.
The reason is that  you mentioned you plug 4 camera into one host
therefore, please ensure you use a powered HUB to provide sufficient power for all cameras
without sufficient power, it also has chance some cameras among them will lost enumeration

RealSenseCustomerSupport, 2020-04-17:
----------------------------------------------

**If we don’t hear from you in 7 days, this issue will be closed**

closed 2020-05-04 by RealSenseCustomerSupport