realsenseai/librealsense #5671: Merging trealsense code rs-save to disk & rs-trajectory

Shivani1796, 2020-01-18:
Hii...
I am combining two realsense code, i.e rs-save-to-disk & rs-trajectory but when I run them together it gives error message : 

`RealSense error calling rs2_pipeline_start_with_config(pipe:0x2844720, config:0x284ca80):
    No device connected
`

ev-mp, 2020-01-19:
@Shivani1796 hello, `rs-trajectory` implies that you're using T265. 
Please fill in the issue template essentials: Device type, SDK and FW versions, OS type required for support. Also - post the code snippet for review.

As for T265 - It could be related to the constrain on how to instante multiple handles to device within the single executable, depending on the SDK version used. #3734, #4363

Shivani1796, 2020-01-19:
@ev-mp Hi.. Thanks for the reply,
I am using T265 realsense camera, Operating system : Linux , SDK Version : 2.31.0
Here is the below code , merged (rs-trajectory & rs-save-to-disk ) 

`// In this example, we show how to track the camera's motion using a T265 device
int main(int argc, char * argv[]) try
{

    // Declare depth colorizer for pretty visualization of depth data
    rs2::colorizer color_map;
    // Declare RealSense pipeline, encapsulating the actual device and sensors
   // rs2::pipeline pipe_for_save_to_disk; 
    // Start streaming with default recommended configuration
   // pipe_for_save_to_disk.start();  
    int imgCount = 0;
    
    
    // Initialize window for rendering
    window app(1280, 720, "RealSense Trajectory Example");

    // Construct an object to manage view state
    glfw_state app_state(0.0, 0.0);
    // Register callbacks to allow manipulation of the view state
    register_glfw_callbacks(app, app_state);

    // Create objects for rendering the camera, the trajectory and the split screen
    camera_renderer cam_renderer;
    tracker tracker;
    split_screen_renderer screen_renderer(app.width(), app.height(), tracker, cam_renderer);
    
    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    
    // Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    
    // Add pose stream
    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
    
    // Start pipeline with chosen configuration
    pipe.start(cfg);
     
    std::ofstream myfile("/home/shivani/Desktop/librealsense-development/build/examples/trajectory/matrix1.txt");
     
    // Main loop
    while (app)
    {
        //HERE RS-SAVE-TO-DISK STARTS ...!!!!!!! 
	// Capture 30 frames to give autoexposure, etc. a chance to settle
	    for (auto i = 0; i < 30; ++i)  pipe.wait_for_frames();

	    // Wait for the next set of frames from the camera. Now that autoexposure, etc.
	    // has settled, we will write these to disk
	 for (auto&& frame : pipe.wait_for_frames())
            {
                
		// We can only save video frames as pngs, so we skip the rest
		if (auto vf = frame.as<rs2::video_frame>())
		{
		
		    auto stream = frame.get_profile().stream_type();
		    // Use the colorizer to get an rgb image for the depth stream
		    if (vf.is<rs2::depth_frame>()) vf = color_map.process(frame);

		    // Write images to disk
		    std::stringstream png_file;
		    png_file << "rs-save-to-disk-output-" << imgCount++ << ".png";
                    
		    stbi_write_png(png_file.str().c_str(), vf.get_width(), vf.get_height(),
		                   vf.get_bytes_per_pixel(), vf.get_data(), vf.get_stride_in_bytes());
		    std::cout << "Saved " << png_file.str() << std::endl;
		}
	    }

        //  RS-SAVE-TO-DISK ENDS ..!!!!!!!! 
       //     RS- TRAJECTORY BEGINS
        // Wait for the next set of frames from the camera
        auto frames = pipe.wait_for_frames();
        std::cout << frames  << std::endl;
        // Get a frame from the pose stream
        auto f = frames.first_or_default(RS2_STREAM_POSE);
        // Cast the frame to pose_frame and get its data
        auto pose_data = f.as<rs2::pose_frame>().get_pose_data();
        float r[16];
        // Calculate current transformation matrix
        tracker.calc_transform(pose_data, r);
        if (myfile.is_open())
	{
	 std::cout << "going to write in matrix file "<< imgCount++ << std::endl;
	 myfile << "";
	 for (int count = 0 ; count <15; count++)
            {
	      myfile << r[count] << " " ;
	    }
	myfile << "\n";
	
	}
	else std::cout<< "unable to open file";
        // From the matrix we found, get the new location point
        rs2_vector tr{ r[12], r[13], r[14] };
        // Create a new point to be added to the trajectory
        tracked_point p{ tr , pose_data.tracker_confidence };
        // Register the new point
        tracker.add_to_trajectory(p);
        // Draw the trajectory from different perspectives
        screen_renderer.draw_windows(app.width(), app.height(), app_state, r);
       
  
    }
    myfile.close();
    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

ev-mp, 2020-01-19:
@Shivani1796 , the code that tries to save files to disk is not consistent:
1.The stream is configured to produce `pose` frames only:
>
```c++
// Create a configuration for configuring the pipeline with a non default profile
rs2::config cfg;
// Add pose stream
cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
// Start pipeline with chosen configuration
pipe.start(cfg);
...
```

while the code tries to generate PNG for video frames, which are not part of the configuration:
>
```c++
 // We can only save video frames as pngs, so we skip the rest
  if (auto vf = frame.as<rs2::video_frame>())
  {
...
```

So this section is never entered/executed resulting in zero PNG files created.

2.To clarify a certain sentence:
> I am using single pipeline called pipe and in case of rs-save-to-disk I do not need to start() with cfg...but in case of rs-trajectory in need to start(cfg) with cfg.

In both cases a `cfg` object is used, but for `rs-save-to-disk` a **_default_** `cfg` is internally generated. So it activates the streams predetermined by the SDK developers:). In the latter case `cfg` object is created and populated externally by **_user-defined_** custom configuration (pose only).

3.Also note that the main loop should be refactored : - the initial auto-exposure settle routine is within the `while(1)` which means that for every single pose frames there is a penalty of 30 unused frames.

Shivani1796, 2020-01-19:
@ev-mp Thanks for quick response..!!!
 We have internal cfg for save to disk which works fine and user defined cfg for trajectory which also works fine..but now i need to  have both code in single file so that both functionalities gets covered...so how I should use pipeline start to handle both functionalities.
Can we have common "cfg" for both of them? So that both can work ..?

Shivani1796, 2020-01-19:
@ev-mp  hiii...
Want to add in my previous comment, what configuration I should use to make it work both for trajectory and save to disk...
Also what I explored is that I cannot use two separate pipeline with user defined cfg and internal cfg for trajectory and save to disk respectively for same device/camera.
So please can you help me in this.?``

ev-mp, 2020-01-20:
@Shivani1796 hello,
I think that the internal cfg covers both use-cases - for T265 the default configuration is Pose+Gyro+Accel+FishEye1/2, and with USB3 and modern CPU there should be no bandwidth or CPU issues to handle.
As for using separate pipelines it is 
a) not necessary as a single pipe will provide all the data required both by save-to-disk and rs-trajectory, and 
b) not possible as the most granular unit that a pipe can handle is a sensor, and T265 comprises of a single sensor (capable of producing the 5 live streams). 

As for handling the 5 streams simultaneously - the easiest way to start with the synchronous API (as it is now) and just iterate over the frames obtained with `pipe.wait_for_frames()`. 
But this can produce artifacts based on the processing time (I/O operations) so performance-wise you should consider moving to asynchronous API (sensor or pipeline callbacks) to minimize data latency and frame drops.
For the same reason I'd recommend to review the requirements and preemptively activate only the streams required by the actual use-case:
```c++
cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
cfg.enable_stream(RS2_STREAM_INFRARED, RS2_FORMAT_Y8,...);
cfg.enable_stream(...
```

closed 2020-07-06 by dorodnic

zainmehdi, 2020-07-07:
@Shivani1796 were you able to solve this issue ?