realsenseai/librealsense #5672: How to use Multiple Pipeline for single device ?????

raghav123456, 2020-01-18:
>  I need to merge rs-trajectory and rs-save to disk program also need to have multiple pipeline single device .

`I have camera/device connected and camera is on in video mode....now i want to have image of that video which i can do using rs-save-to-disk code...also i need to find trajectory of the same video which i can find using rs-trajectory code.`
**Now i want to merge this two code...problem i am facing is rs-trajectory uses cfg when we start pipeline....but rs-save-to-disk does not use cfg when we start pipeline....how can i do this??.....What I found is that I cannot use two pipelines for same camera/device I guess, please help me** 

Thanks in advance....!!!!!!!!

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

You can see what streams are supported for the camera in use (`rs-enumerate-devices -d`), and start one pipeline with the desire configuration.
Thank you.

RealSenseCustomerSupport, 2020-02-01:
----------------------------------------------

Closing this ticket if there is no other question for this topic. Thank you.

closed 2020-02-06 by RealSenseCustomerSupport