realsenseai/librealsense #5676: Realsense-viewer does not recognize SR305

LiranAz, 2020-01-19:
| Required Info:
|---------------------------------|------------------------------------------- |
| Camera Model               | SR300 |
| Operating System & Version | {Linux (Ubuntu 18) |
| Kernel Version (Linux Only) |18.4.3|
| Platform | PC |
| SDK Version                   |2.31.0|
| Language                       | Python |                   

### Issue Description

Issue Description
I installed version 31 as described here:
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
I opened Realsense-viewer and no camera was detected.
[image]
Are there other actions needed to be done? 

Thanks in advance,
Liran.

MartyG-RealSense, 2020-01-19:
The SR305 needs a USB 3 connection.   Have you got the camera plugged into a USB 3 port please?

If you are using USB 3, are you able to find the camera using the Add Source button at the top of the options side panel?

LiranAz, 2020-01-19:
Thanks for the quick response, 
I have it plugged to a USB 3 port.
I cannot find it using the add source button.
[image]

MartyG-RealSense, 2020-01-19:
Could you try running **lsusb** on your Ubuntu please and see whether it can detect the SR305?

LiranAz, 2020-01-19:
It detects nothing that I recognize as the camera.
[image]

MartyG-RealSense, 2020-01-19:
My knowledge of using the SR300 and SR305 on Linux is very limited unfortunately, as all my experience with these models is on Windows.

I have some recollection of past cases that kernel patching was involved in the process.  In the instructions that you linked to, there is a further step after running the RealSense Viewer to validate the installation:

**Verify that the kernel is updated :**
**modinfo uvcvideo | grep "version:" should include realsense string**

The subject is discussed in the link below.

https://github.com/IntelRealSense/librealsense/issues/5229

LiranAz, 2020-01-19:
I tried installing the camera on a windows laptop, same result there.

MartyG-RealSense, 2020-01-19:
On the Windows laptop, are the drivers for the camera visible in the Device Manager window?  The SR300 RealSense drivers will typically be located in the **Imaging Devices** section of the Device Manager, or divided between the **Imaging Devices** section and the **Cameras** section.

LiranAz, 2020-01-20:
I could not locate it there.
[image]

MartyG-RealSense, 2020-01-20:
I believe that if the Windows laptop is connected to the internet then the drivers should download and install automatically when the camera is plugged into the USB 3 port.

If the laptop does not have an internet connection then the drivers can be installed with a Windows program called the Depth Camera Manager (DCM).

https://downloadcenter.intel.com/product/92329/Intel-RealSense-Camera-SR300

LiranAz, 2020-01-20:
The laptop is connected to the internet.
I tried Installing the drivers manually and got the following message:
[image]

MartyG-RealSense, 2020-01-20:
The symptoms that you are getting of the camera not being detected can occur if there is a problem with stability of the power on a particular USB port.  Using a USB hub powered by mains electricity (a 'powered hub') can help with this problem.  

You can also try the camera in a different USB 3 port on a Windows machine if you have more than one USB 3 port.

LiranAz, 2020-01-23:
I tried it again on a windows PC with several USB 3.0 ports.
Same result.
Are there any other things I can try?

MartyG-RealSense, 2020-01-23:
As I mentioned above, a powered USB hub can sometimes help with detection.  If you would rather not buy one without the guarantee of it solving the problem, a free method that has worked for me is to attached a switched off smartphone or tablet to the PC so it begins charging and then insert the camera.  The extra power being supplied to the USB system for device charging may provide the USB power stability required for the camera not to disconnect.

LiranAz, 2020-01-28:
Still does now work, could it be a firmware issue?

MartyG-RealSense, 2020-01-28:
It is difficult to make progress on the problem when neither Windows Device Manager or lsusb can detect the camera at all.

This case was assigned to @RealSenseCustomerSupport so I've tagged their name to help highlight this case to them.

It is rare but possible that the camera hardware is simply broken, in which case you would be entitled to return the camera to where you purchased it from for a refund or replacement if you bought it in the last 30 days.

LiranAz, 2020-01-28:
I understand, hopefully they will be able to help.
Thanks for your efforts.

RealSenseCustomerSupport, 2020-01-28:
----------------------------------------------

Hi @LiranAz,

Please make sure that the USB cable that you use to connect the camera to USB port/hub is in working condition.

Thank you!

LiranAz, 2020-01-30:
Hi,
Thanks for the reply.
The USB is functioning well. I also tried a different one, with no change.

RealSenseCustomerSupport, 2020-01-30:
----------------------------------------------

@LiranAz

Has the camera functioned well/was recognized by OS/Application in the past or it never normally functioned/ was recognized?

LiranAz, 2020-02-02:
No, no device recognizes it.

ev-mp, 2020-02-02:
@LiranAz , when connecting any USB device to Windows PC with speakers should get a distinct sound  that would mean that the device is recognized on the lowerst USB/Power level. Can you confirm this?
Additionally - for the Linux side - can you attach the kernel log produced when the device is plugged-in?

LiranAz, 2020-02-02:
Other devices work on all PCs I tried (linux and windows).
This device does not work on any PC.

I opened syslog and nothing happens when I connect and disconnect the
device.
(I also verified that when I connect another USB device the log is updated
as expected)

ev-mp, 2020-02-02:
@LiranAz,
In case there is no HW activity on the bus during plug-in and you've tried different USB3 cables, then it is safe to assume that the device is bricked and you need to contact the re-seller for RMA replacement.

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

@LiranAz,

Just to make sure,can you please confirm that the camera we are talking about has never normally functioned/ was recognized by OS/Application since the day of the purchase?

Thank you!

LiranAz, 2020-02-05:
Sorry if there was a misunderstanding,
The device worked fine for a few days using SDK version 2.29.0.
Then, immediately after installing version 2.31.0 of the SDK, the device
stopped responding on this or any other computer.

Thanks for your help,
Liran.

RealSenseCustomerSupport, 2020-02-06:
----------------------------------------------

@LiranAz,

Please downgrade the SDK version to 2.29.0 and check whether or not the device is still working
with it.

Thank you!

LiranAz, 2020-02-06:
Tried it with no luck.
As I mentioned above, there was no response.

RealSenseCustomerSupport, 2020-02-06:
----------------------------------------------

@LiranAz,

Did the camera undergo other procedures in addition to SDK upgrade since the day of the purchase  and until the moment it has stopped working? Firmware update, recovery, calibration, etc. - these are the examples of other procedures I meant.

Thank you!

LiranAz, 2020-02-07:
No, nothing at all.

closed 2020-02-09 by LiranAz