realsenseai/librealsense #5680: Save color and depth frame to Bag file ,, Load Bag file

JasonHoon, 2020-01-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                          |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | { D435 } | 
| Firmware Version                    | (05.11.15.00) | 
| Operating System & Version  | { Linux (Ubuntu 18.04)}  | 
| Kernel Version (Linux Only)     | {4.15.0-74-generic}                 | 
| Platform                                  | {PC}  |
| SDK Version                            | { 2.19.1.580 }                          | 
| Language                                | {python }                          | 
| Segment			          | {others }                          | 

### Issue Description

HI 

I'm using 8 depth cameras(D435) on a single PC to save the frames into a bagfiles.

I make 8bagfile( 8cam ).

It's good to save .. but when I tried to load the files, I found a problem.

Color frames are saved well, but the values ​​of some depth frames are not saved.

When I look at the depth image,  I get a black screen.

I can't find similar issues. 

[image]

How can i solve this problem ?

JasonHoon, 2020-01-20:
I connect the cameras directly into the usb3.0 port on my computer. (not usb hub)

JasonHoon, 2020-01-20:
This is spec of my computer

[image]

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

@JasonHoon Did you get black depth frames for streaming only(only preview and no recording)? How did you save the bag files for 8 cameras - save separately or save when streaming together? And how did you load the bag files? Could you please provide the code snippet so that we can further look at it? Thanks!

JasonHoon, 2020-02-04:
@RealSenseCustomerSupport Thank you for response. 
when I saved frames, I can't find problem or error. when I loaded bag file, I got black depth frames for Bag files. I saved the bag files on 8 cameras at the same time.
This is my code for save. 

import pyrealsense2 as rs
import numpy as np
import cv2
import copy
import time 

prevTime = 0
width2 = 424
height2 = 240

frame_fp = 60

depth_image66 = np.zeros((424,240), np.float)

def mouse_callback(event, x, y, flags, param):
	global depth_image6
	print(depth_image66[y,x])

cv2.namedWindow("depth_image_out6")
cv2.setMouseCallback("depth_image_out6", mouse_callback)

pipeline = rs.pipeline()
config = rs.config()
config.enable_record_to_file('Input/GAIT/cam0.bag')         
config.enable_device('828112073069')    # 828112071201   828112073069
config.enable_stream(rs.stream.depth, width2, height2, rs.format.z16, frame_fp)
config.enable_stream(rs.stream.color,  width2, height2, rs.format.bgr8, frame_fp)

pipeline2 = rs.pipeline()
config2 = rs.config()
config2.enable_record_to_file('Input/GAIT/cam1.bag')
config2.enable_device('827312073386')
config2.enable_stream(rs.stream.depth, width2, height2, rs.format.z16, frame_fp)
config2.enable_stream(rs.stream.color,  width2, height2, rs.format.bgr8, frame_fp)

pipeline3 = rs.pipeline()
config3 = rs.config()
config3.enable_record_to_file('Input/GAIT/cam2.bag')
config3.enable_device('828112073054')
config3.enable_stream(rs.stream.depth, width2, height2, rs.format.z16, frame_fp)
config3.enable_stream(rs.stream.color,  width2, height2, rs.format.bgr8, frame_fp)

pipeline4 = rs.pipeline()
config4 = rs.config()
config4.enable_record_to_file('Input/GAIT/cam3.bag')
config4.enable_device('827312072713')
config4.enable_stream(rs.stream.depth, width2, height2, rs.format.z16, frame_fp)
config4.enable_stream(rs.stream.color,  width2, height2, rs.format.bgr8, frame_fp)

pipeline5 = rs.pipeline()
config5 = rs.config()
config5.enable_record_to_file('Input/GAIT/cam4.bag')
config5.enable_device('828112071201') 
config5.enable_stream(rs.stream.depth, width2, height2, rs.format.z16, frame_fp)
config5.enable_stream(rs.stream.color,  width2, height2, rs.format.bgr8, frame_fp)

pipeline6 = rs.pipeline()
config6 = rs.config()
config6.enable_record_to_file('Input/GAIT/cam5.bag')
config6.enable_device('827312072884')  # 828112073079   827312072884
config6.enable_stream(rs.stream.depth, width2, height2, rs.format.z16, frame_fp)
config6.enable_stream(rs.stream.color,  width2, height2, rs.format.bgr8, frame_fp)

pipeline7 = rs.pipeline()
config7 = rs.config()
config7.enable_record_to_file('Input/GAIT/cam6.bag')
[... 347 lines cut ...]
		cv2.putText(depth_colormap, str, (0, 100), cv2.FONT_HERSHEY_SIMPLEX, 1, (255, 0, 0))

		cv2.imshow('depth_image_out', depth_colormap)
		cv2.imshow('depth_image_out2', depth_colormap2)
		cv2.imshow('depth_image_out3', depth_colormap3)
		cv2.imshow('depth_image_out4', depth_colormap4)
		cv2.imshow('depth_image_out5', depth_colormap5)
		cv2.imshow('depth_image_out6', depth_colormap6)
		cv2.imshow('depth_image_out7', depth_colormap7)
		cv2.imshow('depth_image_out8', depth_colormap8)

		cv2.imshow('Test Frame', color_image)
		cv2.imshow('Test Frame2', color_image2)
		cv2.imshow('Test Frame3', color_image3)
		cv2.imshow('Test Frame4', color_image4)
		cv2.imshow('Test Frame5', color_image5)

		cv2.imshow('Test Frame6', color_image6)
		cv2.imshow('Test Frame7', color_image7)
		cv2.imshow('Test Frame8', color_image8)
		cv2.waitKey(1)
		

	#pipeline.stop()
main()

JasonHoon, 2020-02-04:
Actually, I don't need to show depth image when saving frame (cv2.imshow ). 
I need saved Bag Files. I using saved Bag Files(depth image) for Image processing.

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

@JasonHoon Thanks for the quick update! We'll look into your code snippet and update you later. Thanks!

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

@JasonHoon Another thing to check with you is that the dark depth image occurs randomly on all the cameras or only a few cameras of them. Did you get chance to try all the cameras with RealSenseViewer to see if any recording and playback issue?

JasonHoon, 2020-02-05:
@RealSenseCustomerSupport 
At the eight cameras, the dark depth image occurs only on certain cameras(Fourth camera).

So I changed the Fourth camera with a new one, and after testing, I still got the same symptoms with the fourth camera.

And I try all the cameras with RealSenseViewer. recording , playback. 

I could save depth, color image.   there was no dark depth image.

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

@JasonHoon Could you please try the playback for the bag files of forth camera in the RealSenseViewer to see if any dark depth image? Also please try lower FPS(such as 30fps) to see if any helps.

JasonHoon, 2020-02-05:
@RealSenseCustomerSupport  I tried to check the playback for the bag files in the RealSenseViewer.

I couldn't see the dark depth image, but there were not depth value.  The depth value of all pixels in the depth image was zero.

[image]

and I tried to change FPS to 15, 30, 60. I also got dark depth image with the fourth camera.

Actually,  Sometimes it is saved without problems, so there was no dark depth image. 

Problems are found in 5 of 10 attempts.

RealSenseCustomerSupport, 2020-02-06:
----------------------------------------------

@JasonHoon That's really strange. Do you see such zero depth value issue on the remaining cameras in RealSenseViewer when playback the bag files recorded by your code? Do these 8 cameras have the same firmware version?

JasonHoon, 2020-02-06:
@RealSenseCustomerSupport  I think so.. 

Now, I can find such zero depth value on the fourth camera.
When I switch the usb ports connected to the 8 camera, I find the zero depth value on the other camera, not the fourth.

I tried connecting 8 cameras to two USB hubs (4 cameras in one hub), but that wasn't the solution.

Yes All cameras have the same firmware version as 05.11.15.00.

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@JasonHoon Is the camera with zero depth value related to one certain USB port? Or does it happen randomly on all the USB ports?

JasonHoon, 2020-02-13:
@RealSenseCustomerSupport Sorry for the late reply. When I tested, I found some regularity, as you said.

My computer has 4 usb 3.0 ports, and I installed four more (PCI - express ).

so I connect 8 cameras directly into the usb3.0 port on my computer.
port 1 - cam1 
port 2 - cam2 
port 3 - cam3 
port 4 - cam4 
port 5 - cam5 
port 6 - cam6 
port 7 - cam7 
port 8 - cam8 
First, I found problem cam1. so, I changed the ports of cam1 and cam2. 

port 1 - cam2 
port 2 - cam1 
port 3 - cam3 
port 4 - cam4 
port 5 - cam5 
port 6 - cam6 
port 7 - cam7 
port 8 - cam8 
Then I found a problem cam2. 

After several experiments, I I found that the problem occurs on port 1.

but, When I rebooted and tested again, Then I found a problem port2. 

So far I have found problems with port 1 and port 2.

Ports 1 and 2 were originally installed on my computer's motherboard.

One thing I found was that when using the Usb hub, the problem occurred more often.

RealSenseCustomerSupport, 2020-02-18:
----------------------------------------------

@JasonHoon Is it possible for you to try on other host PC to see if get the same issue? Thanks!

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@JasonHoon Did you get chance to try on other host? Looking forward to your reply. Thanks!

JasonHoon, 2020-02-27:
@RealSenseCustomerSupport
 Sorry for the late reply. I did test new computer. 

I was able to record without the problem of having a depth value of zero.

Unfortunately, I found a new error.

This is Error message : RuntimeError: hwmon command 0x30 failed. Error type:  (1).

Thanks! :)

RealSenseCustomerSupport, 2020-02-28:
----------------------------------------------

@JasonHoon Glad to hear the previous issue resolved. For your new error, could you please specify more details about how you get it? And it's better to create another issue if the procedure is not the same as previous one.

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

@JasonHoon As the issue resolved, close it accordingly. Please create another one to track your new issue. Thanks!

closed 2020-03-10 by RealSenseCustomerSupport