realsenseai/librealsense #5683: T265: Attempt to save a map very often ends in "null pointer passed for argument "buffer""

neilyoung, 2020-01-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.908  |
| Operating System & Version |  ubuntu 16.04.6 LTS |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.32.0, development, 95cbea5                       | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

As subject says: What could be the reason for this problem? Comes out of the sudden. I can already have saved the map a couple of times successfully. It can be, that this situation is solved by re-starting. The map is not even corrupted and accepted on load. But there were also cases, in which I had to replug the cam in order to recover from that, IMHO an absolute NOGO for productive scenarios.

radfordi, 2020-01-21:
@neilyoung, please let if know if #5687 addresses this issue.  There's a reasonable chance that it does.

neilyoung, 2020-01-21:
Thanks Jim. This is what I wanted to comment. I have this pretty often, too often, to be honest, to be not concerned while pressing the "Save" button.

Most of the times this is accompanied by a second attempt, which seem to work, but then either stop or a new start does not work and the stack claims "no device connected". The common solution is to power-cycle the cam (not a solution of course :))

radfordi, 2020-01-21:
We have been able to reproduce this.  Thanks for the report, @neilyoung.

neilyoung, 2020-01-21:
Oh great. Thanks for confirmation.

neilyoung, 2020-01-31:
closed for lack of interest

neilyoung, 2020-12-06:
```
2020-12-06 23:07:35,936 pose_pose.py INFO    : SDK: 2.39.0
2020-12-06 23:07:35,936 pose_pose.py INFO    : Firmware: 0.2.0.951
2020-12-06 23:07:35,936 pose_pose.py INFO    : USB: 3.1

```

But still present

closed 2024-04-01 by MartyG-RealSense