realsenseai/librealsense #5684: Pointcloud depth units

nancibles, 2020-01-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description

Hey there! 

My goal is to save a .txt file that contains the depth values corresponding to an RGB frame from the D415. I'm trying to save a pointcloud as follows: 
```
pc = rs.pointcloud()

pc.map_to(color_frames)
points = pc.calculate(frames)

vtx = np.asanyarray(points.get_vertices())
tex = np.asanyarray(points.get_texture_coordinates())

ptcldfile = 'ptcld_{0:0>3}'.format(i)
if os.path.exists(ptcldfile) == False:
    os.mkdir(ptcldfile) 
np.savetxt(ptcldfile + '/pointcloud.txt', vtx, fmt='%.6f',delimiter=',')
```

I've read somewhere that the units are specific to the camera and might not be the actual distance in depth units. I'm a little confused by this since the units pointcloud I'm getting seem to have values that are a little bit off from what I'm expecting given the objects/object distances I have in the scene. I just wanted some clarification on what this pointcloud actually is and what the units are?

dorodnic, 2020-01-21:
The pointcloud is in mm (any depth units are converted prior to this).
If you have high planefit RMS or constant offset, the device may be slightly out of calibration. 
You could try Dynamic Calibration tool from our site, or the experimental Tare and On-Chip calibration options in the Viewer / Depth-Quality Tool

RealSenseCustomerSupport, 2020-01-29:
----------------------------------------------

In addition to @dorodnic suggestion to ensure the camera is properly calibrated, other factors to be considered that could affect depth performance are also discussed here:
https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

Please let us know if there is any other question on the same topic. Thank you.

closed 2020-02-03 by RealSenseCustomerSupport