realsenseai/librealsense #5685: Nodejs example fails to run on macOS

spee, 2020-01-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | Mac  |
| SDK Version                            |  2.31.0 }                          | 
| Language                            |  nodejs                         | 
| Segment			|  others                          | 

### Issue Description

I've cloned the SDK and built from source on macOS. I followed the current version of the installation doc for macOS and I included the node bindings build option. Then I ran `npm install` in both the `nodejs` wrapper folder and its `examples` subfolder. When I try to run `node nodejs-capture.js`, I receive the following error:

```
Error: dlopen([...]/librealsense/wrappers/nodejs/build/Release/node_librealsense.node, 1): Symbol not found: _rs2_delete_sensor_list
  Referenced from: [...]/librealsense/wrappers/nodejs/build/Release/node_librealsense.node
  Expected in: flat namespace
```

Any idea how to fix this? Did I miss a step?

gavan1, 2020-02-20:
@dorodnic let INTEL know all this outdated documentation doesn't help the adoption of it's tools 
https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk#section-building-from-source

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

Hi @spee,

D435i will support on macOS in future release.
Please check the release note.
"""
D435i is supported on Linux, Windows 10 and Android. Mac OS support is scheduled next, and Windows 7 will be provided in future releases. requires FW 5.11.6.250+.
"""

About the dl error, Would you check RPATH in the binary file? or try to use librealsense static librealsense instead.

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

@spee Any other questions about this? Looking forward to your update. Thanks!

spee, 2020-03-31:
@RealSenseCustomerSupport I currently have to focus on other projects so I can't immediately look back into it, sorry!

RealSenseCustomerSupport, 2020-04-27:
----------------------------------------------

@spee Did you get chance to try now? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-05-18:
----------------------------------------------

@spee Any update from your side? Thanks!

spee, 2020-05-18:
No news from me, feel free to close because we aren’t working on this project atm.

closed 2020-05-22 by RealSenseCustomerSupport

shahabl, 2020-08-25:
I'm seeing the exact same issue described here. The examples fail to run with 

`Symbol not found: _rs2_delete_sensor_list`

@RealSenseCustomerSupport, can you re-open this issue and help finding a resolution?