realsenseai/librealsense #5688: Configuration error : RealSense is unable to create streams

Vivoslespagnos, 2020-01-21:
Hi, (I'm French)

So I'm using IntelRealsense_D435 for a project.

My setup :

- Ubuntu 18.04 LTS, Kernel 4.18.0-25-generic (We gonna talk about this kernel version)
- IntelRealsense2 SDK, Firmware 2.31.0 (last)
- Gestoos SDK
- ROS Melodic

The aim is to take gesture detection code from the gestoos SDK and wrapping all of that in ROS for a mobile droîde.

As I work on since 4 months, I know there is only some special kernel for which the DEPTH mode work, it's fot that I'm using the kernel 4.18.0 ;

When I already try lot of fixed solution to unblock te situation, and I have the recurrent error message and it's doesn't work :

crabe@PRJ11:~/PIFE_NOVALES/gestoos/GestoosTESDK_1_5_3/samples/bin/depth/frontal/HandTrackingFrontal$ ./HandTrackingFrontal --interactor_name interactor1.cfg

Enabled streams: DEPTH

Configuration error: RealSense is unable to create streams.

RealSense Error set_pu(id=4) failed! Last Error: Input/output error

Error on rs2_load_json(dev:0x16c6760, json_content:0x16c6ef0, content_size:3451)

RealSense Init FAILED

LOG Failed initializing streams for device : REALSENSE2_D435 mode: DEPTH

Failed initializing depth sensor. Exiting

And when i apply "" ls /dev/vid* "" comand without the camera :

crabe@PRJ11:~$ ls /dev/vid*

/dev/video0 /dev/video1

And when I connect the Realsene_D435 :

crabe@PRJ11:~$ ls /dev/vid*

/dev/video0 /dev/video2 /dev/video4 /dev/video6

/dev/video1 /dev/video3 /dev/video5 /dev/video7

In fact I'm praticaly sure, there is only 3 new way which should appear and not 6 !

So in the shell :

crabe@PRJ11:~$ dmesg | grep uvcvideo

[...]

[ 137.615837] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435 (8086:0b07)

[ 137.618254] uvcvideo: Unable to create debugfs 4-2 directory.

[ 137.618339] uvcvideo 4-6:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!

[ 137.618341] uvcvideo 4-6:1.0: Entity type for entity Processing 2 was not initialized!

[ 137.618343] uvcvideo 4-6:1.0: Entity type for entity Camera 1 was not initialized!

[ 137.618949] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435 (8086:0b07)

[ 137.620706] uvcvideo: Unable to create debugfs 4-2 directory.

[ 137.620881] uvcvideo 4-6:1.3: Entity type for entity Processing 7 was not initialized!

[ 137.620883] uvcvideo 4-6:1.3: Entity type for entity Extension 8 was not initialized!

[ 137.620886] uvcvideo 4-6:1.3: Entity type for entity Camera 6 was not initialized!

[ 177.000652] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp.

[ 177.000652] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).

[ 177.050984] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).

[ 177.101335] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).

[ 177.151666] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).

[ 177.202053] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).

[ 177.252373] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).

[ 177.302731] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).

[ 177.353108] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).

[...]

[...] It's the same thing wich do a loop, I have the impression

[...]

[ 2074.682486] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435 (8086:0b07)

[ 2074.684896] uvcvideo: Unable to create debugfs 4-3 directory.

[ 2074.684965] uvcvideo 4-4:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!

[ 2074.684967] uvcvideo 4-4:1.0: Entity type for entity Processing 2 was not initialized!

[ 2074.684968] uvcvideo 4-4:1.0: Entity type for entity Camera 1 was not initialized!

[ 2074.685549] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435 (8086:0b07)

[ 2074.687272] uvcvideo: Unable to create debugfs 4-3 directory.

[ 2074.687352] uvcvideo 4-4:1.3: Entity type for entity Processing 7 was not initialized!

[ 2074.687354] uvcvideo 4-4:1.3: Entity type for entity Extension 8 was not initialized!

[ 2074.687356] uvcvideo 4-4:1.3: Entity type for entity Camera 6 was not initialized!

[ 2076.585207] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

[ 2076.660731] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

[ 2076.710898] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

[ 2076.781475] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

[ 2076.831664] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

[ 2076.902182] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

[ 2076.952405] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

[...]

[...]

[12527.166776] Modules linked in: hid_generic usbhid hid rfcomm bnep intel_rapl x86_pkg_temp_thermal intel_powerclamp snd_hda_codec_hdmi coretemp snd_hda_codec_idt snd_hda_codec_generic kvm snd_hda_intel irqbypass snd_hda_codec crct10dif_pclmul crc32_pclmul snd_hda_core ghash_clmulni_intel pcbc snd_hwdep aesni_intel wmi_bmof gpio_ich hp_wmi snd_pcm btusb sparse_keymap aes_x86_64 btrtl uvcvideo(OE) crypto_simd videobuf2_vmalloc videobuf2_memops btbcm cryptd snd_seq_midi snd_seq_midi_event i915 videobuf2_v4l2 btintel glue_helper snd_rawmidi videobuf2_common bluetooth drm_kms_helper intel_cstate intel_rapl_perf snd_seq snd_seq_device snd_timer videodev(OE) media drm snd ecdh_generic joydev rtsx_pci_ms memstick soundcore i2c_algo_bit wl(POE) input_leds serio_raw cfg80211 mei_me fb_sys_fops mei ie31200_edac

[12527.166941] uvc_status_cleanup+0x42/0x50 [uvcvideo]

[12527.166944] uvc_delete+0x24/0x170 [uvcvideo]

[12527.166947] uvc_release+0x2f/0x40 [uvcvideo]

Also, when I :

crabe@PRJ11:~$ modinfo uvcvideo | grep "version:"

version: 1.1.2.realsense-1.3.10 (usually, it's ok from this side)

srcversion: 5639FA119ABC88FCED1EA9B

If someone know,

thank you for the time and your help

Vivien

ev-mp, 2020-01-26:
@5693 Vivoslespagnos hello,
I reviewed the attached logs but couldn't identify any setup / hw issues that could result in failures:
- DKMS package is properly installed and loaded. It is confirmed both by the version string and the Kernel log produced by D435 during device enumeration
- You haven't specified what additional UVC devices are additionally attached to host PC, but connecting D435 to the host alone would generate 4 devices (video0..3) - 2 video and 2 (dependent) metadata nodes. This is the expected behavior for kernels 4.16+.
 - The `uvcvideo` kernel messages do not affect functionality and can be disregarded 

The core issue, in my estimation, is within the following section:
>crabe@PRJ11:~/PIFE_NOVALES/gestoos/GestoosTESDK_1_5_3/samples/bin/depth/frontal/HandTrackingFrontal$ ./HandTrackingFrontal --interactor_name interactor1.cfg
Enabled streams: DEPTH
Configuration error: RealSense is unable to create streams.
RealSense Error set_pu(id=4) failed! Last Error: Input/output error
**Error on rs2_load_json(dev:0x16c6760, json_content:0x16c6ef0, content_size:3451)**

It is possible that the json file loaded by the user/3rd party is not compatible/corrupted, and causes the application not to start correctly. If that json file is available then you can try it with `realsense-viewer` or post for review.
I would also recommend to use SDK tools, such as `realsense-viewer`, `rs-capture` and `rs-enumerate-device` to negate HW/Kernel/SDK-related issues.

In any event, since the API call is performed by a 3rd party you need to contact and open a ticket with the developer of `GestoosTESDK_1_5_3` for a tool-specific support.

RealSenseCustomerSupport, 2020-05-03:
----------------------------------------------

Hi.

Will you be needing further help with this? Have you checked in with Gestoos support?

If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-05-11 by RealSenseCustomerSupport