realsenseai/librealsense #5690: How to calculate the extrinsic calibration data for the cameras(T265 and D435) for any Mount?

vnsiot, 2020-01-22:
I wanted to use a 3D printed mount to fix both T265 and D435 cameras (not the one which Intel has provided) and I came across all the threads and examples but didn't find any clue on how to get the extrinsic calibration data.

Thanks,

MartyG-RealSense, 2020-01-22:
A member of the RealSense team wrote a calibration tool to determine the extrinsics between T265 and D4xx via OpenCV and the Python language.

https://github.com/IntelRealSense/librealsense/pull/4355/commits/acb0bbd0efb10ecf81343209d83aeb6cb30e23a2

vnsiot, 2020-01-23:
Thanks for the quick reply, I tried that example which you suggested by mentioning all the arguments (ex. serial numbers, grid height, and width .. etc ) but still it is saying too few arguments is this the right way to calibrate or am I doing it in the wrong way?

Thanks,

MartyG-RealSense, 2020-01-23:
I will tag the creator of the script into this discussion in the hope that they can provide some useful insights about your problem.    @schmidtp1

vnsiot, 2020-01-31:
Hi @schmidtp1, can you please tell me how to use the calibration tool to measure the extrinsic for T265 and D435.

RealSenseSupport, 2021-03-17:
There is the ability to gain the extrinsics of the cameras using the rs-enumerate-devices that provides the extrinsic information of the cameras.  With this information and the mechanical spec of any mount, you'd be able to produce the configuration that is needed.

closed 2021-03-17 by RealSenseSupport