realsenseai/librealsense #5693: Wrong api call exception

BadMachine, 2020-01-22:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |  Win 10 |                                        | 
| Platform                                 | PC |
| SDK Version                            |  Intel® RealSense™ SDK 2.0 (build 2.31.0)                          | 
| Language                            |  C/C++                           | 
| Segment			|  VR                     | 

### Issue Description

When tryin` to create new variable which contains rs2:context class this way:
```C++
rs2::context ctx;
```
Im getting exception in rs_types.hpp:
```C++
            case RS2_EXCEPTION_TYPE_WRONG_API_CALL_SEQUENCE:
                throw wrong_api_call_sequence_error(e);
```

What i tried:
Rebuild SDK from source (no result)
Using pre-built sdk (same)

BadMachine, 2020-01-22:
@ev-mp  Could you please look at this? Thanks!

BadMachine, 2020-01-22:
It seems the problem disappeared when i built RealSense™ SDK 2.0 (build 2.29.0).
Im not sure i should close the issue `cuz original issue about SDK 2.0 (build 2.31.0).

ev-mp, 2020-01-26:
@BadMachine , please attach the code that triggers the error and/or Librealsense logs for review

BadMachine, 2020-01-26:
@ev-mp 
```
auto device = ctx.query_devices();
```
This

ev-mp, 2020-01-27:
@BadMachine , can you post  the code of the main function along with the SDK and kernel logs?
The two lines provided  are taken out of context and therefore of no much help - 
>auto device = ctx.query_devices();
auto dev = device[0];

The only thing I could offer that you instantiate and de-reference an instance from a list without checking whether the list is empty.

BadMachine, 2020-01-29:
Seems the exception appears in:
```C++
rs2::context ctx;
```
Here log:
```
Exception thrown at 0x00007FFD870BA839 in RealMotion.exe: Microsoft C++ exception: librealsense::invalid_value_exception at memory location 0x000000A5844FE3B0.
Exception thrown at 0x00007FFD870BA839 in RealMotion.exe: Microsoft C++ exception: [rethrow] at memory location 0x0000000000000000.
Exception thrown at 0x00007FFD870BA839 in RealMotion.exe: Microsoft C++ exception: rs2::invalid_value_error at memory location 0x000000A5844FE580.
Unhandled exception at 0x00007FFD870BA839 in RealMotion.exe: Microsoft C++ exception: rs2::invalid_value_error at memory location 0x000000A5844FE580.
```

RealSenseCustomerSupport, 2020-05-03:
----------------------------------------------

Hi.

Will you be needing further help with this? Are you still experiencing this with the latest release? And just for sanity are you making sure that the realsense version of the lib you are linking against matches the dll your program is loading at runtime?

If we don’t hear from you in 7 days, this issue will be closed.

Thanks

ev-mp, 2020-05-04:
Closed accordingly

closed 2020-05-04 by ev-mp

billmag, 2020-12-03:
In case anyone else stumbles upon this in the future (hello denvercoder9!), I had this problem crop up when switching from debug to release mode in a Visual Studio project. The issue ended up being that the Release configuration was targeting two separate versions of librealsense.so.

While the whole solution used my vcpkg version of realsense v. 2.39.0 when building in Debug, when I switched to Release, some parts of the project started using an older version of the Realsense SDK that was installed as a Windows package via the regular installer to C:\Program Files<etc>. Uninstalling Realsense via add/remove programs, and then a clean+rebuild in Visual Studio fixed the problem.