realsenseai/librealsense #5696: Improve accuracy of distance measurement and get correct point coordinates

ASSSTUM, 2020-01-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Operating System & Version | Linux (Ubuntu 18.04.3)  | 
| Kernel Version (Linux Only)    |  5.0.0.37                                        | 
| Platform                                 | PC |
| Language                            |  python                         | 
| Segment			|  Object detection and environment detection                     | 

Hello community,

I'm trying to get the point coordinates by given pixel coordinates. The following picture shows the experimental setup.
[image]

The yellow coordinate system is the coordinate system of the camera and the black dashed line is the optical axis (center of the image as well). The red coordinate system is my world coordinate system. The Camera is about 1.28 meters above the ground and it is rotated about 20° around the x-axis. That means, the yellow coordinate system is equal to to red one. Only the origin has been moved in the -y direction and than rotated around the x-axis.
Now I want to estimate the point coordinates of the blue square. This is my code:
```
import sys
import pyrealsense2 as rs
import numpy as np
import cv2
import argparse
import pandas as pd

#pipeline
pipeline = rs.pipeline()
config = rs.config()

#config

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)
config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)

# start and wait for frames
pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        ir1_frame = frames.get_infrared_frame(1)
        ir2_frame = frames.get_infrared_frame(2)

        if not depth_frame or not color_frame:
            sys.stderr.write("Faled to create color frame or depth frame. Quitting.\n")
            sys.exit(1)

        # Intrinsics & Extrinsics
        depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics
        color_intrin = color_frame.profile.as_video_stream_profile().intrinsics
        depth_to_color_extrin = depth_frame.profile.get_extrinsics_to(color_frame.profile)
        color_to_depth_extrin = color_frame.profile.get_extrinsics_to(depth_frame.profile)

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        cv2.circle(color_image, (int(color_intrin.ppx), int(color_intrin.ppy)), 5, (0, 255, 0), 2)
        ir1_image = np.asanyarray(ir1_frame.get_data())
        ir2_image = np.asanyarray(ir2_frame.get_data())

        # Show images
        cv2.namedWindow('RealSense-RGB', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense-RGB', color_image)
        cv2.namedWindow('RealSense-Depth', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense-Depth', depth_image)
        cv2.namedWindow('IR1', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('IR1', ir1_image)
        cv2.namedWindow('IR2', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('IR2', ir2_image)

        #cv2.waitKey(1)
        c = cv2.waitKey(1)
        if c == 27:
          break
        if c == 109: #ASCII 'm'
            x_pix = input('X-Value: ')
            y_pix = input('Y-Value: ')
            print('Pixel Coordinates: x = %s, y = %s'%(x_pix, y_pix))

            # Align depth frame to color frame
            colorizer = rs.colorizer()
            align = rs.align(rs.stream.color)
            frameset = align.process(frames)
            aligned_depth_frame = frameset.get_depth_frame()
            colorized_depth = np.asanyarray(colorizer.colorize(aligned_depth_frame).get_data())
            cv2.circle(color_image, (int(color_intrin.ppx), int(color_intrin.ppy)), 5, (0, 255, 0), 2)
            aligned_2_img = np.hstack((color_image, colorized_depth))
            cv2.imshow('Alignment Depth to Color', aligned_2_img)

            # depth of aligned depth frame
            depth_in_meters = aligned_depth_frame.get_distance(int(x_pix), int(y_pix))
            print('Depth: ', depth_in_meters)

            #intrinsic of the aligned depth frame (should be equal to the color intrinsic matrix)
            aligned_depth_intrin = aligned_depth_frame.profile.as_video_stream_profile().intrinsics

           depth_point = rs.rs2_deproject_pixel_to_point(aligned_depth_intrin,[int(x_pix),int(y_pix)], depth_in_meters)
            print("Camera Coordinate System: x=%.3lf, y=%.3lf, z=%.3lf\n" % (depth_point[0], depth_point[1], depth_point[2]))

            rot = 20.4418
            high = 1.230
            v_cam = np.array([[depth_point[0]],[depth_point[1]],[depth_point[2]]])
            R_world_crane = np.array([[1,0,0],[0,np.cos(np.deg2rad(rot)),np.sin(np.deg2rad(rot))],[0,- np.sin(np.deg2rad(rot)), np.cos(np.deg2rad(rot))]])
            trans = [[0],[-high],[0]]
            v_temp = np.dot(R_world_crane,v_cam)
            v_word = v_temp+trans

            print('\n Vector world:\n', v_word)

finally:

    # Stop streaming
    pipeline.stop()

```
For this code example, please assume, that I know the correct coordinates. That is the reason, why I include the coordinates with the input command.
For the following coordinates, I get this output:
   >X-Value: 106
   >Y-Value: 347
   >Pixel Coordinates: x = 106, y = 347
   >Depth:  2.330000162124634
   >3D-Pos in World: -0.80461, 0.39058, 2.33000

The first value 'x = -0.80461' meters is correct. The depth is a little bit inaccurate because the value is 2.330 meters instead of 2.566 meters. My Problem is the y-value: y = 0.39058. It is absolutely false because the correct value is 1.3 meters.
I assume the error occurs due to the rotation around x. 

I have done some tests. The following diagram shows distance measurements. You can see, that the real distance (legend on the right side) is very different from the depth value of the camera.
[image]

So these are my questions.
1. Is there a way to improve the depth accuracy?
2. How can I get the right y-Value?

Thank's a lot for your help!

MartyG-RealSense, 2020-01-23:
In regard to your first question, you could improve accuracy by increasing the resolution from 640x480 to the D435's optimum resolution of 848x480.

Section 1 (second paragraph of page 1) of Intel's camera tuning guide comments on this subject and advises on settings to use if you have to use a lower resolution than the stated optimum,

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@ASSSTUM For your 1st question, besides the suggestion from MartyG-Realsense to use resolution of 848x480, please also try high accuracy preset to see if helps.

For 2nd question, please move below two lines before "while True:" to get depth to color temporal alignment.
            colorizer = rs.colorizer()
            align = rs.align(rs.stream.color)

RealSenseCustomerSupport, 2020-02-10:
----------------------------------------------

@ASSSTUM Any update from your side?  Thanks!

ASSSTUM, 2020-02-11:
Many thanks for all your support. I will test it as soon as possible. Than I will give you a feedback.

ASSSTUM, 2020-02-13:
I have changed my code according to your suggestions:
```

import sys
import pyrealsense2 as rs
import numpy as np
import cv2

#pipeline
pipeline = rs.pipeline()
config = rs.config()

#config
config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 60)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 60)

# start and wait for frames
cfg=pipeline.start(config)
dev = cfg.get_device()

depth_sensor = dev.first_depth_sensor()
depth_sensor.set_option(rs.option.visual_preset, 3)#set high accuracy: https://github.com/IntelRealSense/librealsense/issues/2577#issuecomment-432137634
depth_sensor.set_option(rs.option.laser_power,359)#[0-359]

colorizer = rs.colorizer()
colorizer.set_option(rs.option.max_distance,12)#[0-16]
align = rs.align(rs.stream.color)
try:
    #print settings:
    print("Visual Preset: %s --> Meaning of value: %s"%(depth_sensor.get_option(rs.option.visual_preset),depth_sensor.get_option_value_description(rs.option.visual_preset, int(depth_sensor.get_option(rs.option.visual_preset)))))
    print("Laser Power: %s --> Valid range: %s" % (depth_sensor.get_option(rs.option.laser_power),depth_sensor.get_option_range(rs.option.laser_power)))
    print("Max Distance: %s --> Valid range: %s" % (colorizer.get_option(rs.option.max_distance), colorizer.get_option_range(rs.option.max_distance)))
    while True:
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()

        if not depth_frame or not color_frame:
            sys.stderr.write("Faled to create color frame or depth frame. Quitting.\n")
            sys.exit(1)

        # Intrinsics & Extrinsics
        depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics
        color_intrin = color_frame.profile.as_video_stream_profile().intrinsics
        depth_to_color_extrin = depth_frame.profile.get_extrinsics_to(color_frame.profile)
        color_to_depth_extrin = color_frame.profile.get_extrinsics_to(depth_frame.profile)

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        #cv2.circle(color_image, (int(color_intrin.ppx), int(color_intrin.ppy)), 5, (0, 0, 255), 2)#image center

        # Show images
        cv2.namedWindow('RealSense-RGB', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense-RGB', color_image)
        cv2.namedWindow('RealSense-Depth', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense-Depth', depth_image)

        c = cv2.waitKey(1)
        if c == 27:
          break
        if c == 109: #ASCII 'm'
            x_pix = input('X-Value: ')
            y_pix = input('Y-Value: ')
            print('Pixel Coordinates: x = %s, y = %s'%(x_pix, y_pix))

            # Align depth frame to color frame
            frameset = align.process(frames)
            aligned_depth_frame = frameset.get_depth_frame()
            colorized_depth = np.asanyarray(colorizer.colorize(aligned_depth_frame).get_data())
            cv2.circle(color_image, (int(color_intrin.ppx), int(color_intrin.ppy)), 5, (0, 255, 0), 2)
            aligned_2_img = np.hstack((color_image, colorized_depth))
            cv2.imshow('Alignment Depth to Color', aligned_2_img)

            # depth of aligned depth frame
            depth_in_meters = aligned_depth_frame.get_distance(int(x_pix), int(y_pix))
            print('Depth: ', depth_in_meters)

            #intrinsic of the aligned depth frame (should be equal to the color intrinsic matrix)
            aligned_depth_intrin = aligned_depth_frame.profile.as_video_stream_profile().intrinsics

            depth_point = rs.rs2_deproject_pixel_to_point(aligned_depth_intrin,[int(x_pix),int(y_pix)], depth_in_meters)
            print("Camera Coordinate System: x=%.3lf, y=%.3lf, z=%.3lf\n" % (depth_point[0], depth_point[1], depth_point[2]))

            neig = 0#20.4418
            high = 1.47
            v_cam = np.array([[depth_point[0]],[depth_point[1]],[depth_point[2]]])
            R_world_crane = np.array([[1,0,0],[0,np.cos(np.deg2rad(neig)),np.sin(np.deg2rad(neig))],[0,- np.sin(np.deg2rad(neig)), np.cos(np.deg2rad(neig))]])
            trans = [[0],[-high],[0]]
            v_temp = np.dot(R_world_crane,v_cam)
            v_word = v_temp+trans
            print('\n Vector Camera:\n', v_cam)
            print('\n Vector rotated:\n', v_temp)
            print('\n Vector world:\n', v_word)
finally:
    # Stop streaming
    pipeline.stop()

```
At the config part I changed the resolution to `config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 60)`. Then I define the settings of the camera:
With this code `depth_sensor.set_option(rs.option.visual_preset, 3)` I try high accuracy preset and with this line `colorizer.set_option(rs.option.max_distance,12)#[0-16]` I make sure, that the max distance is set right.
I will print all settings in the try loop to make sure that all properties are correct.
Unfortunately the depth accurancy will not improve. On top of that, my depth image has lots of black fields.
[image]

Is there another way to improve the accuracy? 
Thanks a lot for your help!

MartyG-RealSense, 2020-02-13:
You may get a better image if you apply post-processing filters.  Intel has a Python tutorial for adding filters that perform a range of image improvement functions including hole-filling.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb

The tutorial shows images 'before' and 'after' applying a range of filters to illustrate the improvement that can be achieved.

**BEFORE**

[image]

**AFTER**

[image]

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@ASSSTUM Any update from your side? Any other questions? Thanks!

RealSenseCustomerSupport, 2020-03-06:
----------------------------------------------

@ASSSTUM Could you please update if you still need support on this?Thanks!

closed 2020-03-06 by ASSSTUM