realsenseai/librealsense #5699: Realsense configuration for use in Lipscan 3D

CamLogan, 2020-01-23:
Hi,

Could someone please help me configure Lipscan 3D with my D435i.

If I export a configuration (i.e. a JSON) from Realsense, can I place it in the folder?

C:\Program Files\WindowsApps\LIPSCorporation.LIPScan_2.2.0.0_x64__7fp2fhkaqcsvm\Opgltest1\Config_files

I would like to use Lipscan to easily create an .obj file but the basic settings are poor.

Thanks

CaM

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-01-23:
As far as I know, LIPS Corp's LIPScan 3D is not linked to the RealSense SDK and so would not be able to make use of a json file created with programs based on the RealSense SDK (such as the RealSense Viewer).

The LIPScan manual says that you can "click upper toolbar Scanner > configure (Ctrl+3) to know the current settings and also for specific alteration based on individual requirements".  Once the scan is taken, you can do further editing on the scan.   this is described in detail from page 19 onwards of the manual, which is linked to below.

https://www.lips-hci.com/download/LIPScan%203D%20-%20User%20Manual-en_0925%20(002).pdf

CamLogan, 2020-01-23:
Hi,

Unfortunately that manual is not related to the LIPS 3D that is downloadable via Microsoft

https://www.microsoft.com/en-us/p/lipscan/9nc85cq8b1p9?rtc=1&activetab=pivot:overviewtab

If I could get that software it would be great, as opposed to the very basic version I downloaded above.

Thanks for your reply in any case.

Cameron

MartyG-RealSense, 2020-01-23:
I do apologise.  I cannot locate an independent download of the version of LIPScan 3D that the linked-to manual referred to.  It seems that it may have been bundled as part of one of LIPS Corp's own-brand depth sensing cameras.

If LIPScan cannot meet your needs, an alternative that is also RealSense compatibkle may be DotProduct's entry-level scanning product Dot3D Scan, which has a 14 day free trial version.  The full version of the software has a subscription fee of $245 a year.

 https://www.dotproduct3d.com/dot3dscan.html

CamLogan, 2020-01-23:
Hi,

Thanks for trying.

CaM

https://www.dotproduct3d.com/dot3dscan.html

RealSenseCustomerSupport, 2020-01-29:
----------------------------------------------

Hi @CamLogan Do you further help on this?

If we don't hear from you in 7 days, this issue will be closed.

closed 2020-02-08 by RealSenseCustomerSupport