realsenseai/librealsense #5701: OpenCV wrapper example not converting from RGB to BGR correctly

cedriclmenard, 2020-01-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA Jetson TX2  |
| SDK Version                            |  v2.31.0                         | 
| Language                            |  C++                          | 
| Segment			|  others                          | 

### Issue Description
When using the example OpenCV wrapper file [cv-helpers.hpp](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/cv-helpers.hpp), the function `frame_to_mat` does not perform a correct conversion from RGB to BGR for OpenCV. This leads to color images used in OpenCV to have swapped blue and red channels.

Between line 25 and 29, the conversion is currently done as such:
```
else if (f.get_profile().format() == RS2_FORMAT_RGB8)
{
    auto r = Mat(Size(w, h), CV_8UC3, (void*)f.get_data(), Mat::AUTO_STEP);
    cvtColor(r, r, COLOR_RGB2BGR);
    return r;
}
```

It should be done as such:
```
else if (f.get_profile().format() == RS2_FORMAT_RGB8)
{
    auto r_rgb = Mat(Size(w, h), CV_8UC3, (void*)f.get_data(), Mat::AUTO_STEP);
    Mat r_bgr;
    cvtColor(r_rgb, r_bgr, COLOR_RGB2BGR);
    return r_bgr;
}
```

I will try and make a pull request.

Thank you.

cedriclmenard, 2020-01-23:
Made a PR, #5702 to address this issue.

dorodnic, 2020-02-17:
Thanks @cedriclmenard

closed 2020-02-28 by RealSenseCustomerSupport