realsenseai/librealsense #5704: Announcement:  RealSense SDK version 2.32.1 now available (massive update)

MartyG-RealSense, 2020-01-23:
Hi everyone,

RealSense SDK version 2.32.1 is now available, featuring a huge list of significant additions and bug fixes.  

https://github.com/IntelRealSense/librealsense/releases/tag/v2.32.1 

This includes a new software infrastructure and a new firmware version for T265 that adds exciting features and fixes many T265 problems on MacOS.

Also of interest to MacOS users will be an update of the installation instructions for use of Librealsense with Catalina.

The full release notes can be found in the link below:

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2321

yputra, 2020-01-25:
I tried the new firmware on my Raspberry Pi 4 to run the T265 together with the D435. I started the T265 first and then the D435 with delayed roslaunch in ROS. It does not show any error and the device seems to run well, but when I stream the data from T265, it actually stopped giving output (pose, etc.) after few minutes. It seems like it crashed but the software think it is still fine.

MartyG-RealSense, 2020-01-25:
Have you tried the launch file for running a D4xx camera and T265 together please?

 **roslaunch realsense2_camera rs_d400_and_t265.launch**

yputra, 2020-01-25:
> Have you tried the launch file for running a D4xx camera and T265 together please?
> 
> **roslaunch realsense2_camera rs_d400_and_t265.launch**

Yes, I just did, still the same problem. The ROS launch did not show any error or termination, but I dont receive anymore data after few minutes

MartyG-RealSense, 2020-01-25:
If you are connecting to the D435 and T265 using device serial number, an alternative approach to launching may be to launch by USB port ID instead.  This method was created after a number of users experienced problems with the D435 and T265 combination when using serial numbers to distinguish between the devices.  I believe this avoids the need for using launch delay timing.

https://github.com/IntelRealSense/realsense-ros#launch-parameters

closed 2020-01-29 by MartyG-RealSense