realsenseai/librealsense #5707: running error of Intel.RealSense.Viewer.exe

bokkngo, 2020-01-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

bokkngo, 2020-01-24:
I could not open the Intel.RealSense.Viewer.exe file. I'd like to test my Sense 3D scanner SR300.
My notebook is totally covering the hardware recommendations:
i5 7300HQ (so 7th gen.) 2.5 GHz
Geforce GTX 1050Ti Graphic card
RAM: 8 GB
1920 x 1080 screen resolution
900 GB free space

I've tried run the program on 3 different ways:
Simple doubleclick
right mouse/open
right mouse + run as administrator

Thanks in advance,
Adam

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

Hi Bokkngo,
what is the OS and version you are using now? windows 10
please check if the viewer is minimized into a small icon, that you will need to maximized it

Attachment(s):
maximize it.jpg - https://realsensesupport.intel.com/attachments/token/nc3DhhBwzFspMxhPdHJyiXyZ5/?name=maximize+it.jpg

hcostelha, 2020-02-10:
I am having this problem too, and it is driving me nuts. I have tried with the "single viewer" and through the one installed with the SDK. The samples work just fine, including the rs-pointcloud.exe, but both the viewer and the dept-quality applications close immediately after being opened (I do not even get to see a window pop up) . I have tried running through the command line (power shell) but the result is the same and no output is given.

I am using Windows 10 64-bit (1909 (10.0.18363)) with the Intel.RealSense.SDK-WIN10-2.32.1.1297 with the D435 on a ASUS UX430UA.307. I have an Intel® HD Graphics 620 with the drivers 26.20.100.7755 (09/01/2020).

I have confirmed that in my privacy options that the option for apps to use the camera is enabled.

I cannot view the attached show in the previous comment, but I can see in the Task Manager that the realsense-viewer.exe shows starts and the immediately shuts down.

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

please see if below suggestion can help on your case
https://github.com/IntelRealSense/librealsense/issues/5511

hcostelha, 2020-02-24:
> please see if below suggestion can help on your case
> #5511

Thanks for the reply, but I am not sure of what can I try from #5511, given that I am running in Windows, not Linux. If I start a linux guest virtual machine within this Windows host, I can run the viewer. If I start the viewer in Windows, it just closes without ever even showing the viewer window.

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi hcostelha,

able to setup a PC with windows 10 64-bit (1909 (10.0.18363)) with the Intel.RealSense.SDK-WIN10-2.32.1.1297 with the D435 on one Intel NUC  with Intel® Iris Plus Graphic 640..
so the OS version and SDK version is the same..
but unfortunately not able to duplicate the issue.
to run the viewer/depth quality tools, it require 3rd library such as imgui and  glfw to create sophisticated UI, other simple sample may not need them
to get more clue.. we also have the source code for viewer
you can build the solution (which include viewer and all examples) from scratch
and see if issue can be fixed or step run to see where cause issue (ex calling imgui function)

RealSenseCustomerSupport, 2020-03-09:
----------------------------------------------

you can also check if your graphic card/driver support opengl no not
the viewer default use opengl for rendering
basically, the Intel graphic card should support it
you should be able to check this by using tool like GLview

Alnilam1305, 2020-03-11:
I have the same problem. I was able to build to project then with VS and stepped through. 
My problem occurs inside void prepare_config_file() in ux-window.cpp. It seems, that my PC cannot find/resolve the path in 
config_file::instance().set_default(configurations::viewer::log_filename, get_folder_path(special_folder::user_documents) + "librealsense.log");
The program switches then to 
throw std::runtime_error("Failed to get requested special folder");
and finishes. When I comment out the throwing of the error, the program starts and seems to be running. Unfortunately the runtime_error is not showed on the console.

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

Hi Alnilam1305
Thanks for sharing your finding.
basically the viewer will set the default folder to store the log file as
C:\Users\XXXXXX(user name)\Documents\librealsense.log
it looks on your PC, this path is not able to be resolved/find the Documents folder
you can set the default folder to Desktop or others by changing the option to
get_folder_path(special_folder::user_desktop)

RealSenseCustomerSupport, 2020-04-06:
----------------------------------------------

Hi bokkngo,
Had provided suggestion can help on your case?

hcostelha, 2020-04-07:
Hi @Alnilam1305 , that was indeed the problem. In my case, because I have OneDrive installed, there is no `C:\Users\xxxx\Documents`, but a `C:\Users\xxxx\OneDrive\Documents`.
What I did to solve the problem was opening a command prompt (cmd.exe) with administrator privileges and created a symlink to that folder, using the following command in the `C:\Users\xxxx` folder:
`mklink /D "C:\Users\xxxx\OneDrive\Documentos" Documents`

RealSenseCustomerSupport, 2020-04-14:
----------------------------------------------

thanks for confirming the default log folder path and provide the scenario that the default folder would not exist,
We will check this and find a way to prevent this kind of issue

closed 2020-05-14 by RealSenseCustomerSupport