realsenseai/librealsense #5710: Per-frame motion metadata is enabled for Accel but not Gyro

ruffsl, 2020-01-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.20.4-042004-generic                                         | 
| Platform                                 | PC amd64  |
| SDK Version                            |  2.32.1 RELEASE (deb)                         | 
| Segment			|  Robot                          | 

### Issue Description
I'm trying to verify that the clock domain for the accelerometer and gyro are both `Global Time`. However, with current release, it seems that only the gyro has faltered to `System Time`: 

[image]

This error message reports:

> Per-frame metadata is not enabled at the OS level!
Please follow the installation guide for the details.

I presume the message is referencing this document:

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#metadata-support-for-intel-realsense-devices

However, all other metadata streams seem to correctly display a `Global Time` clock domain:

[image]

What could cause this inconsistency? I've been able to replicate the exact issue on two different ubuntu PCs, two different D435i devices, as well as building from source as of 83f952a .

FYI: versions of debian package used: 

```
$ apt list --installed librealsense2*
Listing... Done
librealsense2/bionic,now 2.32.1-0~realsense0.1913 amd64 [installed,automatic]
librealsense2-dkms/bionic,now 1.3.12-0ubuntu1 all [installed]
librealsense2-gl/bionic,now 2.32.1-0~realsense0.1913 amd64 [installed,automatic]
librealsense2-udev-rules/bionic,now 2.32.1-0~realsense0.1913 amd64 [installed,automatic]
librealsense2-utils/bionic,now 2.32.1-0~realsense0.1913 amd64 [installed]
```

Second sample device without latest firmware:

[image]

ev-mp, 2020-01-26:
@ruffsl hello,
The difference in timestamp domains for gyro and accel is rooted in kernel inconsistency and requires kernel patch to rectify. The SDK provides the patches suites for Ubuntu LTS kernels, and v4.20 is not [on the list](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=2&cad=rja&uact=8&ved=2ahUKEwju_eWYgKHnAhVEzKQKHUa9ANgQFjABegQIDBAF&url=https%3A%2F%2Fwww.thomas-krenn.com%2Fen%2Fwiki%2FUbuntu_LTS_Hardware_Enablement_Stack&usg=AOvVaw2Wtxg7ooVUGlzL9J_m4a_C). 
You can switch to LTS kernel and have the kernels patches applied [manually](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation)/[with DKMS](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution) , or, you can port the kernel patches to the specific kernel you're using (imo patches for 4.18 are closest to origin)
Duplicate of #3980

ruffsl, 2020-01-26:
Ok, it wasn't obvious to me that at first, that installation of the DKMS deb packages and manually patching of the LTS kernel where both required simultaneously. I've switched back to the LTS kernel, and applied the kernel patch via the repo provided patch script, however rebooting after the patch, I'm still encountering the same issue.

```
$ sudo apt-get --only-upgrade install librealsense2-utils librealsense2-dkms
[sudo] password for ruffsl: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
librealsense2-dkms is already the newest version (1.3.12-0ubuntu1).
librealsense2-utils is already the newest version (2.32.1-0~realsense0.1913).
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.

$ uname -a
Linux dox 4.15.0-74-generic #84-Ubuntu SMP Thu Dec 19 08:06:28 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

$ realsense-viewer 
 26/01 13:16:41,438 WARNING [140269806679808] (types.cpp:49) set_pu(id=9) failed! Last Error: Input/output error
 26/01 13:16:41,439 WARNING [140269806679808] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 26/01 13:16:56,647 WARNING [140270577784704] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed. Error type: No data to return (-21).
 26/01 13:16:56,866 WARNING [140270577784704] (backend-v4l2.cpp:1208) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 26/01 13:17:06,979 WARNING [140269504673536] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:02.0/0000:3e:00.0/usb4/4-1/4-1:1.5/0003:8086:0B3A.0003/HID-SENSOR-200073.1.auto/iio:device1/buffer/enable
 26/01 13:17:07,061 WARNING [140269471102720] (sensor.cpp:968) HID timestamp not found, switching to Host timestamps.
 ...
```

Log from `scripts/patch-realsense-ubuntu-lts.sh`:

<details>

```
$ ./scripts/patch-realsense-ubuntu-lts.sh 

Remove all RealSense cameras attached. Hit any key when ready

Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version (1.07.1-2).
build-essential is already the newest version (12.4ubuntu1).
git is already the newest version (1:2.17.1-1ubuntu0.5).
linux-headers-generic is already the newest version (4.15.0.74.76).
The following packages were automatically installed and are no longer required:
  libllvm8 libllvm8:i386 libnvidia-common-415
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-bionic-master folder

Package required libelf-dev:  - not found, installing now...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libllvm8 libllvm8:i386 libnvidia-common-415
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  libelf-dev
0 upgraded, 1 newly installed, 0 to remove and 8 not upgraded.
Need to get 57.3 kB of archives.
After this operation, 374 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 libelf-dev amd64 0.170-0.4ubuntu0.1 [57.3 kB]
Fetched 57.3 kB in 0s (156 kB/s)      
Selecting previously unselected package libelf-dev:amd64.
(Reading database ... 640999 files and directories currently installed.)
Preparing to unpack .../libelf-dev_0.170-0.4ubuntu0.1_amd64.deb ...
Unpacking libelf-dev:amd64 (0.170-0.4ubuntu0.1) ...
Setting up libelf-dev:amd64 (0.170-0.4ubuntu0.1) ...
Missing package installed
Package required elfutils:  - not found, installing now...
[... 59 lines cut ...]
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 679 (offset 11 lines).
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 2782 (offset 11 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Hunk #1 succeeded at 1235 (offset 13 lines).
Hunk #2 succeeded at 1273 (offset 13 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 190 (offset 3 lines).
Hunk #2 succeeded at 211 (offset 3 lines).
Applying realsense-hid patch
patching file drivers/iio/accel/hid-sensor-accel-3d.c
Hunk #1 succeeded at 289 (offset 3 lines).
patching file drivers/iio/gyro/hid-sensor-gyro-3d.c
Hunk #3 succeeded at 197 (offset 3 lines).
Hunk #4 succeeded at 213 (offset 3 lines).
Hunk #5 succeeded at 248 (offset 3 lines).
Applying realsense-powerlinefrequency-fix patch
patching file drivers/media/usb/uvc/uvc_ctrl.c
Retrofit uvc bug fix enabled with 4.18+
patching file drivers/media/usb/uvc/uvc_video.c
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/media/usb/uvc/uvc_video.c.rej
```

ruffsl, 2020-01-26:
After looking closer at the `patch-realsense-ubuntu-lts.sh` script, looks like it stalls from the `v1-media-uvcvideo-mark-buffer-error-where-overflow.patch` may already be upstreamed in Ubuntu LTS tag Ubuntu-4.15.0-74.84. I commented this section out, and proceed to install the built kernel modules:

https://github.com/IntelRealSense/librealsense/blob/83f952a4bd6b70d72459f66c7f67ddaba9d337a0/scripts/patch-realsense-ubuntu-lts.sh#L131-L134

(We may want to update that if condition logic.)

<details>

```
$ ./scripts/patch-realsense-ubuntu-lts.sh

Remove all RealSense cameras attached. Hit any key when ready

Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version (1.07.1-2).
build-essential is already the newest version (12.4ubuntu1).
git is already the newest version (1:2.17.1-1ubuntu0.5).
linux-headers-generic is already the newest version (4.15.0.74.76).
The following packages were automatically installed and are no longer required:
  libllvm8 libllvm8:i386 libnvidia-common-415
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-bionic-master folder

Package required libelf-dev:  - found
Package required elfutils:  - found
Package required bison:  - found
Package required flex:  - found
Fetching Ubuntu LTS tag Ubuntu-4.15.0-74.84  to the local kernel sources folder
The kernel has modified files:
	modified:   drivers/iio/accel/hid-sensor-accel-3d.c
	modified:   drivers/iio/gyro/hid-sensor-gyro-3d.c
	modified:   drivers/media/usb/uvc/Makefile
	modified:   drivers/media/usb/uvc/uvc_ctrl.c
	modified:   drivers/media/usb/uvc/uvc_driver.c
	modified:   drivers/media/usb/uvc/uvc_video.c
	modified:   drivers/media/usb/uvc/uvcvideo.h
	modified:   drivers/media/v4l2-core/v4l2-ioctl.c
	modified:   include/uapi/linux/videodev2.h
Proceeding will reset all local kernel changes. Press 'n' within 3 seconds to abort the operation
Do you want to proceed? [Y/n]
Resetting local changes in ubuntu-bionic-master folder
 HEAD is now at e1afa08ac478 UBUNTU: Ubuntu-4.15.0-74.84
Switching to LTS tag Ubuntu-4.15.0-74.84
HEAD is now at e1afa08ac478 UBUNTU: Ubuntu-4.15.0-74.84
Applying patches for bionic-master line
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file drivers/media/v4l2-core/v4l2-ioctl.c
Hunk #1 succeeded at 1248 (offset 2 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 succeeded at 679 (offset 11 lines).
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 2782 (offset 11 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Hunk #1 succeeded at 1235 (offset 13 lines).
Hunk #2 succeeded at 1273 (offset 13 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 190 (offset 3 lines).
Hunk #2 succeeded at 211 (offset 3 lines).
Applying realsense-hid patch
patching file drivers/iio/accel/hid-sensor-accel-3d.c
Hunk #1 succeeded at 289 (offset 3 lines).
[... 366 lines cut ...]
  LD [M]  /home/ubuntu/intel/librealsense/ubuntu-bionic-master/drivers/media/v4l2-core/v4l2-fwnode.ko
  LD [M]  /home/ubuntu/intel/librealsense/ubuntu-bionic-master/drivers/media/v4l2-core/videobuf-dma-sg.ko
  LD [M]  /home/ubuntu/intel/librealsense/ubuntu-bionic-master/drivers/media/v4l2-core/videobuf2-core.ko
  LD [M]  /home/ubuntu/intel/librealsense/ubuntu-bionic-master/drivers/media/v4l2-core/videobuf2-dvb.ko
make: Leaving directory '/home/ubuntu/intel/librealsense/ubuntu-bionic-master'
Patched kernels modules were created successfully

0
Replacing videodev  -
	Applying the patched module ...  succeeded
Replacing uvcvideo  -
	Applying the patched module ...  succeeded
Replacing hid_sensor_accel_3d  -
	Applying the patched module ...  succeeded
Replacing hid_sensor_gyro_3d  -
	Applying the patched module ...  succeeded

Script has completed. Please consult the installation guide for further instruction.
```

</details>

Looks like the gyro metadata is working fine now:

[image]

ev-mp, 2020-01-27:
@ruffsl hello,it great that you've managed to resolve the issue.
Two notes:
- There is no need to apply both DKMS and the manual script. Actually it must be avoided as the methods are functionally-equivalent, so running both may lead to unexpected results. You need to choose and implement one of the methods as prescribed. DKMS should be preferred for LTS kernels.
- The interrupt for user input with 4.15.0-70+ will be addressed to allow for unattended patching.

ruffsl, 2020-01-27:
>There is no need to apply both DKMS and the manual script. ... DKMS should be preferred for LTS kernels.

Thanks for the clarification. After I had rolled back to the LST kennel with the DKMS still installed, the issue persisted until I manually patched. Perhaps if I happen to repeat this with a fresh Ubuntu install this wouldn't be repeatable, but in theory the DKMS provided should work with 4.15.0-74.84 ?

ev-mp, 2020-02-04:
The DKMS package for Ubuntu 18 is compatible with LTS kernels 4.15 and 4.18 branches (all patch numbers included).
It will be extended to support 5.0 and 5.3 in the next release.

ruffsl, 2020-02-05:
> It will be extended to support 5.0 and 5.3 in the next release.

Nice, looking forward to simply using the DKMS package for the current kernel shipped in Ubuntu.

I've noticed that for `5.0.0-23-generic`, I encountered device detection issues when applying the 3rd-party patchs that affects USB2 behavior, and so got working again by commenting them out.

https://github.com/IntelRealSense/librealsense/blob/83f952a4bd6b70d72459f66c7f67ddaba9d337a0/scripts/patch-realsense-ubuntu-lts.sh#L129-L146

closed 2020-02-05 by ruffsl