realsenseai/librealsense #5713: How to avoid memory leak and run program forever

OneQuestion, 2020-01-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

OneQuestion, 2020-01-25:
I'am aware of #2860 memory leak problem. Is there a way to avoid it by restarting service once in a while? I tried pipe.stop() and then pipe.start() again, but it made no difference.
My code goes like this:

`pipe.start()`
`frames = pipe.wait_for_frames();`
`depth = frames.get_depth_frame();`
`points = pc.calculate(depth);`
`vertices = points.get_vertices();`
`tex_coords = points.get_texture_coordinates();`
`pipe.stop();`

and repeat

MartyG-RealSense, 2020-01-25:
Librealsense has a script instruction called **hardware_reset()** that can reset the camera automatically once a certain condition is met.  For example, one user set its activation to be conditional on a certain number of zero-value pixels being detected.

A simple C++ reset script for a single camera is provided in the link below.

https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

If multiple cameras are connected, the reset script below iterates through all connected cameras:

https://forums.intel.com/s/question/0D70P0000068r7PSAQ/d435-streams-work-once-then-crash?language=en_US

OneQuestion, 2020-01-25:
so **hardware_reset()** should be used instead of pipe.stop() ? Does it have same effect as exiting the program?

MartyG-RealSense, 2020-01-25:
pipe.stop stops the currently active camera streams that were initiated by starting the pipe with pipe.start.   

A hardware_reset() is essentially the same as resetting the camera by doing a physical unplug and replug of the camera's USB cable to disconnect and reconnect it.

OneQuestion, 2020-01-25:
so I shouldn't use stop or start in this situation at all? only these ? :
`	        rs2::context ctx;`
        `    rs2::device dev = ctx.query_devices().front();`
          `  dev.hardware_reset();`
          `  rs2::device_hub hub(ctx);`
         `   dev = hub.wait_for_device();`

OneQuestion, 2020-01-25:
but it gives me **Frame didn't arrive within 15000**

MartyG-RealSense, 2020-01-25:
Well, you need pipe.start of course to start the streams running in the first place.

Using hardware_reset is a common way to deal with problems that may arise during a program designed for long-running over a number of days. 

 RealSense 400 Series cameras can run indefinitely so long as they remain within operating temperature tolerances.  When a problem with a long-run occurs, it is typically related to a non-camera issue such as the USB port / cable, computer operating system or computer hardware.  The outer limits for running a program without problems occurring tends to be 4 days.  So doing a hardware reset once every 24 hours, for example, could help reduce the chances of a problem occurring.  Industrial-grade USB and computer equipment may also help to increase long-run reliability.

Stopping a pipeline properly is important because you can introduce problems if a shutdown of a RealSense program is done improperly.  So pipe.stop does certainly have an important role to play.  It is up to you to program the conditions under which pipe.stop will be allowed to occur.
   
Regarding the frame didn't arrive error: what FPS speed are you running the program at please?

OneQuestion, 2020-01-25:
default, so I guess 30fps
this is how I get this error:

    dev = ctx.query_devices().front();
    dev.hardware_reset();
    dev = hub.wait_for_device();
    pipe.stop();
    pipe.start();
    frames = pipe.wait_for_frames();
    depth = frames.get_depth_frame();
    points = pc.calculate(depth);
    vertices = points.get_vertices();
    tex_coords = points.get_texture_coordinates();`
also, it constantly gives me this error until I physically reconnect camera:
**terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x10 failed. Error type:  (1).**

MartyG-RealSense, 2020-01-25:
pipe.stop normally goes at the end of the script, not before pipe.start. 

As I am not a RealSense programming specialist, there are people more able on this forum to give expert advice on this topic.   In the meantime, it may be a good idea to take a look at the SDK's point cloud example program.  Once started, it generates the point cloud in a window and keeps running until the window is manually closed by clicking on its corner-cross.   This sounds like the kind of project that you have in mind.

https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud

closed 2020-12-05 by OneQuestion