realsenseai/librealsense #5715: T.265: SVG available

neilyoung, 2020-01-26:
For a software documentation (especially for illustrating the diverse coordinate system transformations) I would need an SVG of the T.265. Is that available somehow?

MartyG-RealSense, 2020-01-26:
If you mean Scalable Vector Graphics, I have not personally come across an image of the T265 in that format.  A CAD file of the T265 in STP format is available though from the link below.

https://dev.intelrealsense.com/docs/cad-files

There is also a mechanical line-drawing of the T265 on page 28 of the data sheet document for the T265.

https://dev.intelrealsense.com/docs/tracking-camera-t265-datasheet

neilyoung, 2020-01-27:
@MartyG-RealSense Thanks, yes, I meant SVG.  Maybe I can convert from STP to SVG. Thanks

closed 2020-01-28 by neilyoung