realsenseai/librealsense #5717: Firmware  5.12.2.100 causes realsense-viewer to crash when imu is on D435i

Ezward, 2020-01-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.02.100| 
| Operating System & Version |  Ubuntu18.04  | 
| Kernel Version (Linux Only)    |  5.3.0-26-generic | 
| Platform                                 | MSI Gaming Laptop RTX2060  |
| SDK Version                            |  2.32.1 | 
| Language                            | realsense-viewer | 
| Segment			| Robot-donkeycar | 

### Issue Description
I updated SDK, then updated firmware. Now when I run realsense viewer, if I use rgb and depth everything is fine (although missng 848x480 resolution, which I used to have), but if I turn on Motion Module, realsense-view crashes.  Restarting and it will crash as well, indicating camera is in a bad state.  Hardware reset from realsense-view did not work.  Uplugging camera, quitting realsense-viewer, plug back in camera and restart realsense-viewer and I'm back to where I can use RGB and Depth.

ev-mp, 2020-01-27:
@Ezward hello, I tried D435i with the same FW and kernel versions as specified but could not reproduce any of the above in `realsense-viewer`/ `rs-capture` / `rs-measure`.
Can you turn on and provide the SDK and kernel logs for review ?

Additionally I'd recommend to try a different USB cable/port as some of the warnings are associated with power/hw setup.

Ezward, 2020-01-27:
It was the cable.  I was using a 3rd party cable when getting the errors.  When I switch to the supplied cable, everything worked.  Thank you.

closed 2020-01-27 by Ezward