realsenseai/librealsense #5718: `depth_frame` pixel format

merttoka, 2020-01-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  NA | 
| Platform                                 | PC  |
| SDK Version                            |  2.31                          | 
| Language                            |  C++                           | 
| Segment			|  Network                          | 

### Issue Description
I am using C++ and I am trying to serialize aligned frames along with following properties: _serial number_, _lastFrameTime_, _width_, _height_, _colorImage_, and _depthImage_. I, then, compress and transmit the buffer over the network. On the receiving end, I uncompress and deserialize the object. Unfortunately I am stuck with deserializing depth frame: 

```c++

size_t Serialize() {
	// this->msgBuffer is allocated with 4MB

	rs2::video_frame color = frames[c_id];
	rs2::depth_frame depth = frames[d_id];
	
	size_t w = color.get_width();
	size_t h = color.get_height();

        // do we really need two functions?
	auto _memcopy = [](char* buf, size_t& offset, auto item, size_t length) {
		memcpy(buf + offset, reinterpret_cast<char*>(&item), length);
		offset += length;
	};
	auto _memcopy_char = [](char* buf, size_t& offset, char* item, size_t length) {
		memcpy(buf + offset, item, length);
		offset += length;
	};

	size_t lastIndex = 0;
	// 12 bytes serial number
	_memcopy_char(msgBuffer, lastIndex, (char*)serial_number.c_str(), strlen(serial_number.c_str()));
	//  8 bytes time
	_memcopy(msgBuffer, lastIndex, lastFrameTime, sizeof(lastFrameTime));
	//  4 bytes height
	_memcopy(msgBuffer, lastIndex, h, sizeof(h));
	//  4 bytes width
	_memcopy(msgBuffer, lastIndex, w, sizeof(w));
	//  4 bytes width -- needs repeating for some reason
	_memcopy(msgBuffer, lastIndex, w, sizeof(w));
	// w*h*{bytes p. pix.} bytes
	_memcopy_char(msgBuffer, lastIndex, (char*)color.get_data(), color.get_data_size());
	
        // Crashes Here
        _memcopy_char(msgBuffer, lastIndex, (char*)depth.get_data(), depth.get_data_size());

	return lastIndex;
}
```
_(Excuse my ugly code)_
Code above runs if the type of my depth frame is either `rs2::frame` or `rs2::video_frame` instead of `rs2::depth_frame`. When I define `depth` as `rs2::depth_frame`, program throws an exception on `depth.get_data()`:
```
RealSense error calling rs2_get_frame_data(frame_ref:nullptr):
    null pointer passed for argument "frame_ref"
```

When I define it as `rs2::frame`, `.get_data()` and `.get_data_size()` functions work without an exception, however I am confused about the format of my depth image. `.get_data_size()` returns a number equal to `width*height*3`, meaning 3 bytes per pixel, but I don't know the correct format that can deserialize this? In another issue [#5546], I've seen that depth images use `Z16` format, and I believe it means a `short` (2 bytes). 

In this case, how many bytes of the buffer belong to first pixel value of my depth image, and what's the type of the value?

merttoka, 2020-01-27:
Now I am experiencing a similar problem with `rs2::points` type:

```c++
rs2::points p = frames_per_stream[points_id]; // points from filtered depth frame
std::cout << p.size() << " " << p.get_data_size() << " " << sizeof(rs2::vertex) << std::endl;
// prints
// 19200 384000 12
```

As I understand the results, I have 19200 vertices, each containing 20 bytes of data. However `sizeof(rs2::vertex)` returns 12 (as expected). What other information is stored on the `rs2::points` object?

merttoka, 2020-02-04:
Found out that the structure of the memory block returned by `p.get_data()` contains elements in this order:
```
coord_0, ..., coord_n, uv_0, ..., uv_n
```
not 
```
coord_0, uv_0, coord_1, uv_1, ..., coord_n, uv_n
```

`n` being the linear index of `rs2::points p`.

closed 2020-02-04 by merttoka

netanelfar, 2020-08-25:
@merttoka i am using python but how did you get the full size of the frame? is there any thing i need to change when before using depth_frame.get_data_size() ?

merttoka, 2020-08-25:
@netanelfar I haven't used the Python API, I am afraid I don't know if there is anything specific to `pyrealsense`. Perhaps this issue could be useful: https://github.com/IntelRealSense/librealsense/issues/6271#issuecomment-616354511