realsenseai/librealsense #5720: RealSenseViewer hangs at exit

laguerreche, 2020-01-27:
| | |
|---|---|
|**librealsense**|2.32.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense T265|
|**Serial Number**|0000852212110652|
|**Firmware Version**|0.2.0.926|
|**Usb Type Descriptor**|3.1|
|**Product Line**|T200|

RealSenseViewer hangs when I try to quit by clicking on the close button of the window.

Steps to reproduce:
- starts the viewer with a connected camera,
- click on the toggle button on the "tracking module" line to turn on the camera,
- click on the usual right-top close icon of Windows to exit: you will then have to kill the app to close it.

Sometimes it works... A race condition seems to be there... I attach a [debug log](https://github.com/IntelRealSense/librealsense/files/4118026/log-debug-realsenseviewer.txt).

laguerreche, 2020-01-27:
A behavior that may be related...

If I run the sample "rs-capture" in VS 2017 (from the same SDK release built in x86_64), it hangs when I close its viewer window. If I then pause the debugger, I can see that the callstack of the main thread is blocked in the `concurrency.h` file at line 303:
`_was_flushed_cv.wait_for(lock_was_flushed, std::chrono::hours(999999), [&]() { return _was_flushed.load(); });`

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

Hi @laguerreche

Thanks for the submission and information.

Is this causing a significant issue for you in what you're trying to accomplish?

neilyoung, 2020-02-05:
I noticed a similar behavior on Windows 10 Home with 2.32.1 from PyPi. It was the stop (pipe.stop()) which hangs. But I was just making a quick test, not doing anything with probably patched USB drivers. The issue was reproducible.

laguerreche, 2020-02-06:
@RealSenseCustomerSupport, my app lets the user start or stop tracking at any time because there's no need to run tracking all the time.

With the bug I reported, it means that the user cannot stop tracking because the app will freeze. When exiting the app, the app freezes if the user started tracking and then the app must be killed.

So it's quite a significant issue... As a workaround, I remain on 2.30.0 (which crashes in C# when iterating the disconnected devices from the "Context.OnDevicesChanged" event).

closed 2020-12-08 by RealSenseSupport