realsenseai/librealsense #5721: Adding fisheye distortion model in C# wrapper

laguerreche, 2020-01-27:
In SDK 2.32.1, the C# wrapper still doesn't expose the fisheye Kannala Brandt distortion model. Could you add it to `Distortion.cs` file? In the enum, it would take value 5.

Example:
```
namespace Intel.RealSense
{
    /// <summary>
    /// Distortion model: defines how pixel coordinates should be mapped to sensor coordinates.
    /// </summary>
    public enum Distortion
    {
        /// <summary> Rectilinear images. No distortion compensation required.</summary>
        None = 0,

        /// <summary> Equivalent to Brown-Conrady distortion, except that tangential distortion is applied to radially distorted points</summary>
        ModifiedBrownConrady = 1,

        /// <summary> Equivalent to Brown-Conrady distortion, except undistorts image instead of distorting it</summary>
        InverseBrownConrady = 2,

        /// <summary> F-Theta fish-eye distortion model</summary>
        Ftheta = 3,

        /// <summary> Unmodified Brown-Conrady distortion model</summary>
        BrownConrady = 4,

        FisheyeKannalaBrandt = 5
    }
}
```

RealSenseCustomerSupport, 2020-02-15:
----------------------------------------------

Thank you for your contribution.

There is the PR. https://github.com/IntelRealSense/librealsense/pull/5853

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@laguerreche PR#5853 has been merged to the latest v2.34. Please have a check. Thanks!

https://github.com/IntelRealSense/librealsense/pull/5853

RealSenseCustomerSupport, 2020-04-15:
----------------------------------------------

@laguerreche Did you get chance to try? Looking forward to your update. Thanks!

closed 2020-04-26 by dorodnic